WHILL/mrp-support #4: バッテリーについて

Hiroo43, 2021-04-21:


KazumichiShirai, 2021-04-28:
時速6km/hで走行した時の仕様となります。

時速によって走行距離はほとんど変わらないです。

Hiroo43, 2021-04-28:
回答ありがとうございます。
もう少し教えてください。

この質問をさせていただいた背景は
フル充電で連続走行させたときに何時間駆動するかを知るためでした。

時速によって変化が少ないということであれば
単純に何時間動かすことが可能でしょうか。

以上、よろしくお願いいたします。

KazumichiShirai, 2021-04-28:
理論上は例えば時速1kmなら16時間連続稼働します。長時間になると、システム電流の影響も見えてくるので、16時間よりは短いと思います。
ただし、
・減速/加速のカーブ
・減速/加速の回数
・路面
・坂
・WHILL model CRから他のセンサやPCに電力を供給しているか否か
で駆動時間は大きく変わるので、ご注意ください。

Hiroo43, 2021-04-28:
繰り返しになってしまいますが、速度で走行距離が大きく変わらないということは
時速1kmなら16時間
↓
時速4kmなら4時間
ということでは無いということでしょうか。
時速が何kmであろうと16時間くらい稼働するということで合っていますでしょうか。

以上、よろしくお願いいたします。

KazumichiShirai, 2021-04-30:
時速1kmなら16時間
↓
時速4kmなら4時間
となります。走行条件によるので、概算となることはご注意ください。

>時速が何kmであろうと16時間くらい稼働するということで合っていますでしょうか。
違います。16時間は時速1kmの場合です。

closed 2021-06-28 by github-actions[bot]