WHILL/mrp-support #5: ROSを用いた自律移動型電動車いすの開発

Takehiro-nlab, 2021-05-13:
御社の製品にPCとVLP-16(Velodyne Lidar)を搭載し、自律移動型電動車いすを開発をしようと考えています。
購入に際しまして、いくつかお聞きしたいことがございます。

1．公開されているパッケージのうち必要なパッケージはros_whillだけですか？

2．Gazeboでのシミュレーション時のような独自のコントローラを設けなくとも、cmd_velにメッセージを送信するだけで実機は動作しますか？

3．ジョイスティック操作に関して得られる情報はどのようなものですか？

4．連続的なジョイスティック操作によるマニュアル走行は可能ですか？

5．ジョイスティックの操作情報をもとにcmd_velにメッセージを送信する独自のインターフェース設計を施す場合、どのようなトピック間の通信を行えばいいですか？

以上、よろしくお願いいたします。
詳細に教えていただけると幸いです。

asmsuechan, 2021-05-13:
おもむろにissue巡りをしていて偶然目に入ったのでコメントします。(私はWHILLさんの人ではありません)

> 1．公開されているパッケージのうち必要なパッケージはros_whillだけですか？

動かすだけならros_whillだけで動きます。が、自律移動となるとmove_baseなり何なり別のパッケージを使う方が楽だと思います。↓は以前書いたlaunchファイルの中身です。
https://github.com/morioka-lab/whill_navigation/blob/main/launch/whill_navigation.launch

> 2．Gazeboでのシミュレーション時のような独自のコントローラを設けなくとも、cmd_velにメッセージを送信するだけで実機は動作しますか？

cmd_velなトピックを投げてやるだけで動きました。トピックの投げ先は正確には/whill/controller/cmd_velですが。

> 4．連続的なジョイスティック操作によるマニュアル走行は可能ですか？

やればできるはずです！

## 他
電動車椅子くんをシリアル通信できるよう改造するのが少し大変そうでした(私はやってない)。何かいいやり方あるのだろうか。

## refs
私の出身研究室のHPの記事
[車椅子ロボットWHILLをROSで自律走行させる](https://moriokalab.com/news/103)

実際に自律移動してみた様子はこちらの動画です。レーザーセンサーはHokuyoの2DLiDARです。
https://mobile.twitter.com/suenaga_ryota/status/1346036870413602818

s-katsu, 2021-05-14:
@asmsuechan コメントありがとうございます。ご回答いただいた内容で相違ありません。

@Takehiro-nlab 
WHILLの勝又と申します。お問い合わせありがとうございます。

基本 @asmsuechan さんのコメントの通りですが、一部補足させていただきます。

> 1．公開されているパッケージのうち必要なパッケージはros_whillだけですか？

rosはros_whillのみです。他のライブラリは本repoのREADMEに記載しています。

https://github.com/WHILL/Model_CR_Technical_Support#libraries

> 3．ジョイスティック操作に関して得られる情報はどのようなものですか？

rosとして機体側から出力する方向でpublishされるtopicはros_whillのREADMEご参照ください。

https://github.com/WHILL/ros_whill#published-topics

> 4．連続的なジョイスティック操作によるマニュアル走行は可能ですか？

可能です。

> 5．ジョイスティックの操作情報をもとにcmd_velにメッセージを送信する独自のインターフェース設計を施す場合、どのようなトピック間の通信を行えばいいですか？

ros_whillとしましては、READMEに記載の通りI/Fはrostopicで定義されており、このケースですと~states/joyをsubして、cmd_velをpubするros nodeを用意していただくことになります。

Takehiro-nlab, 2021-05-15:
@asmsuechan 
@s-katsu 
ご回答ありがとうございます！
続けてご質問させてください。

1．トピックとしてのodomがpubされている他、パラメータ設定によってodomからbase_linkへのtfをpubできる思いますが、これらは搭載されたimuの情報を用いて補正されたものですか？

s-katsu, 2021-05-18:
@Takehiro-nlab 
> トピックとしてのodomがpubされている他、パラメータ設定によってodomからbase_linkへのtfをpubできる思いますが、これらはWhillに搭載されたimuの情報を用いて補正されたものですか？

こちらはimuによって補正はしていません。純粋にモーターの回転を見ているエンコーダーの値から算出されたodomになります。

Takehiro-nlab, 2021-05-18:
ありがとうございます！

closed 2021-05-18 by Takehiro-nlab