WHILL/mrp-support #6: WHILL Model C/CKについて

Yoshioka592, 2021-05-18:
WHILL Model C/CKについてお聞きしたいのですが、
ヴイストン株式会社様が販売しているVS-C3というコントローラでWHILL Model C/CKを動かすことは可能でしょうか。
もし、動かすことが可能なら教えていただけないでしょうか。

s-katsu, 2021-05-19:
お問い合わせありがとうございます。

まずモデル名についてですが、Model C/CKは外部から制御可能なモデルではなく、一般的な方が使用されるパーソナルモビリティとなります。
外部から制御可能な研究開発モデルはModel CRになります。

https://whill.inc/jp/model-cr

こちらのモデルは外部と通信可能なRS232Cの通信ケーブルが出ていますので、それとPCなどを接続してシリアル通信で制御いただくかたちとなっております。

https://www.vstone.co.jp/products/vs_c3/index.html
VS-C3はこちらの製品かと思いますが、そちらの受信モジュールがどういったプロトコルで制御するか次第となります。
申し訳ありませんが、こちらは他社製品となり、そちらがどういった制御をされているかはWHILLからはお答え出来かねます。
少なくともWHILL Model CRではVS-C3のサポート対応は行っておりません。

弊社のModel CRとしましては、通信仕様および、制御ライブラリはこちらに公開しておりますので、そちらをご参照ください。

https://github.com/WHILL/Model_CR_Technical_Support#communication-specifications

https://github.com/WHILL/Model_CR_Technical_Support#libraries

Yoshioka592, 2021-05-19:
回答をしていただきありがとうございます。

WHILL Model C/CKと申しましたがWHILL Model CRでした申し訳ございません。
続けて質問をして申し訳ないのですがWHILL Model CR自体に元々付いているコントローラは他の外部との移動ロボットに接続することが可能なんでしょうか。
WHILL Model CRのコントローラはWHILL Model CRを動かすためだけのコントローラなのでしょうか。
難しい質問をして申し訳ないのですがご回答いただけると幸いです。

closed 2021-06-28 by github-actions[bot]