WHILL/mrp-support #8: 入力の周期

Takehiro-nlab, 2021-06-09:
ジョイスティックからの入力値は、どのぐらいの周期で得られますか？

s-katsu, 2021-06-11:
お問い合わせありがとうございます。

Model CRのシリアル通信の仕様としましては、
Model CR -> 外部へ送信されるデータの周期はStartSendingData Commandのinterval of sending dataで決定されます。（最小10ms)

[通信仕様書からの抜粋]
[image]

https://github.com/WHILL/whill_control_system_protocol_specification/blob/master/WHILL_Control_System_Protocl_Specification.pdf

ですのでジョイスティックの入力値が得られる周期もinterval of sending dataの設定値によって決まります。

もしrosなどのライブラリを使用される場合はそちらの内部で設定されているinterval of sending dataの設定値に依存しますので、ご使用のライブラリがありましたら教えていただけると回答いたします。

Takehiro-nlab, 2021-06-11:
ros_whillを使わせていただこうと考えています。

s-katsu, 2021-06-22:
> ros_whillを使わせていただこうと考えています。

ros_whillの場合はModel CRからの受信周期は20ms間隔となります。

https://github.com/WHILL/ros_whill/blob/melodic-devel/src/ros_whill.cpp#L528

Takehiro-nlab, 2021-06-22:
ありがとうございます！

closed 2021-06-22 by Takehiro-nlab