WHILL/mrp-support #11: 電動車いすを自律走行させるときのスピードについて

mayo233, 2021-07-09:
現在、rviz上にある2D nav goalで目的地を指定してから電動車いすを自律走行させています
しかし、自律走行中、車椅子のスピードが出すぎているため落としたいです
なお、ros_whillパッケージにあるinitial_speedprofile.yamlの値を変えてもだめでした
また、move_baseノードを起動させるとき、<param name="controller_frequency" value="1.5" /と設定しています。controller_frequency valueの値を落としたら車椅子が滑らかに動きません。ところが、この値を上げすぎると車椅子が早く動いて危険です
どうすれば周波数が高く、スピードが遅い状態で車椅子を自律走行できるでしょうか？

s-katsu, 2021-07-12:
ros_whillにpubしているtopicは
controller/joy
controller/cmd_vel
のどちらですか？

cmd_velの場合は0.004km/h単位で速度指令値を設定することになるため、設定する値で速度を調整いただく必要があります。
https://github.com/WHILL/ros_whill/blob/melodic-devel/src/whill/whill_commands.cpp#L117

mayo233, 2021-07-12:
使用しているコマンドはcontroller/cmd_velです

mayo233, 2021-07-12:
後，https://github.com/WHILL/ros_whill/blob/melodic-devel/src/whill/whill_commands.cpp#L117を起動させないとスピード調整が出来ないのでしょうか？

s-katsu, 2021-07-12:
> 後，https://github.com/WHILL/ros_whill/blob/melodic-devel/src/whill/whill_commands.cpp#L117を起動させないとスピード調整が出来ないのでしょうか？

申し訳ありませんが、こちらの質問がよく理解出来ておりません。

ros_whillを起動してもらえば/whill/cmd_velをsubscribeして、指定値に従って制御可能です。
cmd_velの場合は速度指令値は0.004km/h単位でしか指定出来ません。

cmd_velは速度指令値を設定するros topicになりますので、そこで指令値を調整することで速度を調整することは可能です。
http://wiki.ros.org/ja/ROS/Tutorials/UnderstandingTopics

closed 2021-08-18 by github-actions[bot]