WHILL/mrp-support #15: joystickの無効化

Takehiro-nlab, 2021-07-21:
/whill/states/joyのデータをもとに独自の/whill/controller/cmd_velを送信するモジュールを作成しました。whillの電源をONにするとrosとの接続の有無に関わらずジョイスティック操作が可能ですが、この接続を断ち切りたいです(ジョイスティックのデータは受け取りたい)。この場合の対処法をお教えいただけませんか？

s-katsu, 2021-07-28:
機体のジョイスティックを無効にしたい場合は、
ros_whillのコードの下記の箇所を0->1に変更することで可能です。変更後にbuildすれば反映されるはずです。

https://github.com/WHILL/ros_whill/blob/836b2faf1d8b0c9d8f2034f431a048287bd187f4/src/whill/whill_commands.cpp#L75

詳細は通信仕様書 [WHILL_Control_System_Protocl_Specification.pdf](https://github.com/WHILL/whill_control_system_protocol_specification/blob/master/WHILL_Control_System_Protocl_Specification.pdf)のSet Joystick Commandに記載されています。

[image]

Takehiro-nlab, 2021-08-03:
0x00を0x01に変更しビルドしましたが、起動後ジョイスティックによるマニュアル操作が可能なままでした。
ご確認をお願いいたします。

s-katsu, 2021-08-12:
@Takehiro-nlab  
大変申し訳ありませんが、私の前回の回答が間違っていました。

もう一度通信仕様書の内容を確認しますと、
> Command ID: 0x03 
> U0:Disable/Enable joystick control of user/host device.
> U0 = 0:Disable joystick control of user and enable joystick control of the host device
> U0 = 1:Enable joystick control of user and disablejoystick control of the host device

となっておりますので、U0 = 0で本体のジョイスティックが無効で外部（host device)のジョイスティックが有効となります。
ですのでros_whillは特に変更不要でした。大変失礼致しました。

実際にModel CRのPCを接続、そのPCにJoystick（Game Pad)をつないだ状態で確認しましたが、
/whill/controller/joyがpubされている間は、本体のジョイスティックが無効になっていることを確認しました。

もし常に本体のジョイスティックを無効にされたい場合は常に/whill/controller/joyをpublishされている状態にすれば実現可能です。

ちなみにテストで使用した"/whill/controller/joy"をpubするjoy_nodeのlaunchファイルはこのようになっています。
```
<launch>
    <node name="joynode" pkg="joy" type="joy_node" args="" output="screen" required="true">
        <param name="autorepeat_rate" value="100"/>
        <remap from="/joy" to="/whill/controller/joy" />
        <param name="dev" value="$(optenv JOY_WHILL /dev/input/js0)"/>
    </node>  
</launch>
```

Takehiro-nlab, 2021-08-20:
/whill/controller/joyに仮想のジョイスティックからメッセージをpubすることでユーザー操作を無効化することができました。
/whill/states/joyは変わらずsubできたので問題ありません。
ありがとうございました。

s-katsu, 2021-08-23:
@Takehiro-nlab 無効化出来てよかったです。
解決済みでしたら、IssueのCloseをお願いします。お問い合わせありがとうございました。

closed 2021-08-23 by Takehiro-nlab