WHILL/mrp-support #16: バッテリーの温度条件について

asakamo, 2021-07-28:
ModeCRのバッテリーの温度条件は-15℃～+40℃となっていますが、夏場の屋外や寒冷地など、
この範囲外となってしまう場所で使用した場合どのようなことが想定されるでしょうか。
安全装置が働き出力を停止するなど想定される動作がありましたら教えていただけないでしょうか。

よろしくお願いいたします。

s-katsu, 2021-07-28:
>ModeCRのバッテリーの温度条件は-15℃～+40℃となっていますが、夏場の屋外や寒冷地など、
この範囲外となってしまう場所で使用した場合どのようなことが想定されるでしょうか。

いろいろな条件があるため簡単に説明することは出来ないですが、走行出来ない、そもそも使用不可と考えていただいた方がよろしいです。

asakamo, 2021-08-02:
ご回答ありがとうございます。温度条件範囲外で使用不可であること承知いたしました。
お手数をおかけして申し訳ありませんが、追加で３点質問させてください。
・走行中にバッテリーの雰囲気温度が温度条件範囲を超え高温になってしまった場合、
　安全装置が働き出力が停止するでしょうか。
・安全装置が働く場合はその温度は何度になるでしょうか。
・保管時バッテリーの雰囲気温度が60℃付近になった場合発煙発火の恐れはないでしょうか。

よろしくお願いいたします。

s-katsu, 2021-08-03:
恐れ入りますがバッテリーの使用に関しては取扱説明書に記載されている以外の情報はお答え出来かねます。
バッテリーに関しては前回回答しました通り、温度、バッテリー容量、負荷状況（走行している環境）に依存してどういった安全機能が働くか異なりますので簡単にお答えすることは出来ません。

申し訳ありませんが、説明書の使用条件をご確認いただき、使用可能な範囲内でのご使用をお願いします。

asakamo, 2021-08-03:
回答しづらい質問を申し訳ありません。
下記認識であっているでしょうか。
・バッテリーには安全機能が存在していて環境に合わせて適切に働く。
・説明書の動作範囲内であれば安全に動作するが、意図せず動作範囲を超えてしまった場合の動作については回答できない。

s-katsu, 2021-08-12:
>下記認識であっているでしょうか。
・バッテリーには安全機能が存在していて環境に合わせて適切に働く。
・説明書の動作範囲内であれば安全に動作するが、意図せず動作範囲を超えてしまった場合の動作については回答できない。

はい、あっています。

asakamo, 2021-08-12:
承知しました。
ご回答ありがとうございました。

closed 2021-08-12 by asakamo