WHILL/mrp-support #17: 電動車いすの速度の追従について

KenEchoVR, 2021-08-15:
現在，自身で指定する入力関数(比例関数やsin関数など)に従って速度を調整したいと考えており，具体的には一定の加速度で速度を加速・減速させたいと考えています．これを実現するために，Python用のSDKを使用し，SetJoystickにより速度を調整できるか行いました．しかし，車いすの速度は入力関数に正しく追従しないという結果となりました．

図１はその時の測定結果であり，SetJoystickのfrontの入力を１秒間隔で１ずつ減らした際の入力数値(青色)と，その時の実際の速度(緑色)をグラフで表したものになります．なお，この時の最高速度は0.5m/s(入力値100)です．グラフを見る限り，入力数値が１下がっても，速度が追従することなく一定のままであり，ある程度入力数値を下げることにより，速度が下がっています．

一応SetJoystickを使用して速度を追従させることができないか模索し，入力数値が変わる際に一度電動車いすに停止信号(入力値0)を送ることで，ある程度追従させることができました．その時の入力数値と実際の速度は図２になります．しかし，ある程度入力関数に追従するようになりましたが，一度停止信号を挟んでいるので，急激な加速度が加わってしまうため，一定の加速度での加速・減速は実装できませんでした．

[image]
図1 入力関数と実際の速度

[image]
図2 停止信号を挟んだ際の入力関数と実際の速度

また，以下のリンクの記事より，SetVelocity(Ros用SDKではSetSpeed)を使用すれば，かなりの精度で指定した速度に追従するとのことなのですが，Python用SDKにはその関数が実装されていませんでしたので，Ros用のSDKを用いて実際に動かしてみました．しかし，電動車いすは動作しませんでした．また，データシートを参考にPython用SDKの方でも実装し実行しましたが同様に動作しませんでした．

この原因としては，現在使用しているWHILLは，記事よりも前に購入したものであり，また，記事にもFWの更新についての記載がありますので，FWが古いのが原因でないかと考えています．

長々と書きましたが，質問したいことは２つになります．
１．SetJoystickでは速度を入力関数に従って追従させる方法がないのか．
２．SetVelocityはどのようにすれば使用できるようになるのか．また，FWの更新が原因の場合，更新してもらうためにはどうすればいいのか．

お忙しいと思いますが，返信をお待ちしております．
また，入力関数に従って速度を追従させる方法が私の考えている方法以外になりましたら，そちらも教えていただけると幸いです．

記事：https://eng-memo.info/blog/cr_new_fw/
データシート：https://github.com/WHILL/whill_control_system_protocol_specification/blob/master/WHILL_Control_System_Protocl_Specification.pdf

s-katsu, 2021-08-16:
ご連絡ありがとうございます。
具体的なグラフまでご提示いただき助かります。

>長々と書きましたが，質問したいことは２つになります．
>１．SetJoystickでは速度を入力関数に従って追従させる方法がないのか．

「SetJoystickでは速度を入力関数に従って追従させる方法」はございません。
SetVelocityを使っていただくかたちでお願いします。

弊社が用意しているライブラリはros_whillしかSetVelocity対応しておりませんが、ご確認いただいたとおり、シリアル通信の仕様で存在していればpythonで実装しても使用可能です。

>２．SetVelocityはどのようにすれば使用できるようになるのか．また，FWの更新が原因の場合，更新してもらうためにはどうすればいいのか．

Model CRのFirmwareが古い場合はSetVelocity対応しておりません。
弊社営業からModel CR購入者様宛にFWの更新の案内がきているかと思いますので、弊社営業までご連絡いただければFWアップデートの対応致します。

KenEchoVR, 2021-08-17:
ご返信ありがとうございます．

質問へのご回答より，SetVelocityコマンドを使用することにいたしました．
そのため，FWの更新をお願いしたいのですが，現在私が使用しているModel CRの購入者(私の所属している研究室の教授)に伺ったところ，FWの更新の案内が来たかわからないと言われました．

従って，もし購入者宛の案内がないとFWの更新ができないのでしたら，再度案内を送っていただきたいと考えており，そのためのやり取りの方法を教えていただければと思います．

また，仮に案内がなくてもFWの更新ができるのでしたらその方法を教えていただければ幸いです．

s-katsu, 2021-08-18:
@KenEchoVR  FWの更新に関しましてはお手数おかけしますが、こちらの問い合わせ窓口からご連絡お願い致します。

https://whill.inc/jp/whill-contact#content-form-corporate-customer-rental

KenEchoVR, 2021-08-22:
@s-katsu ご返信ありがとうございます．
提示していただきました窓口より，FWの更新の手続きをしようと思います．

s-katsu, 2021-10-04:
こちらCloseさせていただきます。何かあれば再Openお願いします。
お問い合わせありがとうございます。

closed 2021-10-04 by s-katsu