WHILL/mrp-support #18: 電動車いすに搭載されている加速度センサーの値とその単位について

KenEchoVR, 2021-08-24:
電動車いすの走行中の振動をデシベル値で求めるために，加速度を求めたいと考えています．

現在Python用SDKを用いて，WHILLに搭載されている加速度センサーの値を取得しているのですが，データシートを見る限り単位が[mg]であるとのことですが，これは，「ミリ・ジー」で１[mg] = 10^(-3)[g] = 9.8 * 10^(-3)[m/s^2]のことでしょうか？

また，静止状態での加速度センサーの値をグラフにしたものが下の図１になります．
図２はデータシートより，加速度センサーの位置と軸を表したものです．
図１より，ｘ軸の加速度は約750，ｙ軸の加速度は約-60，ｚ軸の加速度は約-640となったのですが，なぜ，ｙ軸の(絶対)値が一番大きくならないのですが？
図２を見る限り，ｙ軸が重力方向であるのに，値としては，ｘ軸やｚ軸の加速度のほうが大きくなっているのですがなぜですか？

[image]
図１：静止状態での加速度センサーの値のグラフ
[image]
図２：WHILL搭載の加速度センサーの位置と軸

質問をまとめると，
１．Python用SDKを用いて加速度センサーの値を取得するとき，その値の単位は[mg (ミリ・ジー)]で１[mg] = 10^(-3)[g] = 9.8 * 10^(-3)[m/s^2]であっているでしょうか？
２．静止状態において，センサーの軸方向と重力を加味しても，各軸の加速度センサー値の図２のような値や比にならないのですが，なぜでしょうか？

s-katsu, 2021-08-24:
@KenEchoVR お問い合わせありがとうございます。
大変申し訳ありませんが、Model CR搭載のIMUは精度が低く、ロボティクス的な利用には厳しいため、利用は非推奨とさせていただいております。IMUを利用されたい場合は別途外付けのIMUを購入をご検討お願い致します。
SDKからもIMUのI/Fは削除する予定です。

申し訳ありませんが、よろしくお願いします。

KenEchoVR, 2021-08-24:
@s-katsu ご回答ありがとうございます．
別途外付けのIMUを装着しての利用を考えようと思います．

s-katsu, 2021-10-04:
こちらCloseさせていただきます。何かあれば再Openお願いします。
お問い合わせありがとうございます。

closed 2021-10-04 by s-katsu

sota1111, 2022-10-07:
Model Cのページに横流れ防止機能があると記載があります。
この機能はRS232Cで速度を送信したときも有効なのでしょうか？
CRにも横流れ防止機能はありますか？