WHILL/mrp-support #20: 車輪角度検出センサ、速度検出センサについて

tszkittsang9, 2021-10-02:
Model CRを研究に活用しています。車輪角度検出センサ（速度検出センサ）について質問があります。
(1)Python用SDKにおいて、RIGHT_MOTOR_ANGLE やLEFTT_MOTOR_ANGLE の検出に使っているセンサはどのようなセンサか教えていただけないでしょうか？
(2)RIGHT_MOTOR_SPEEDやLEFT_MOTOR_SPEEDの検出は速度センサを上記センサとは別に持って検出しているのでしょうか？それとも上記角度センサの信号を利用して計算しているのでしょうか？
(3)合わせて、COUNTERとセンサとの関係を教えてください。

KazumichiShirai, 2021-10-05:
ご質問、ありがとうございます。
(1) モーター制御で使いっているセンサを使っております。申し訳ございませんが、詳細は控えさせていただきます。モーターをベクトル制御するのに使っているセンサであり、それなりの精度はあると考えています。
(2)RIGHT_MOTOR_SPEEDやLEFT_MOTOR_SPEEDは上記センサを使って算出した値にLow pass filterをかけております。ですので、自動運転などの速度値に精度を求める場合は、RIGHT_MOTOR_ANGLE /LEFTT_MOTOR_ANGLEを使って算出していただいたほうがよいかと思います。
(3) COUNTERは(1)の角度検出センサがセンシングしたタイミングでインクリメントされます。通信周期の設定やご利用されているOSの都合で、RIGHT_MOTOR_ANGLE /LEFTT_MOTOR_ANGLEだけでモーター回転速度を検出しようとすると、精度が落ちる場合があります。ですので、COUNTER値を使って、モーター回転速度を算出していただくとより高い精度でモーター回転速度を求めることができます。
以下、WHILL Control System Protocol Specification pdfからの抜粋です。ご参照ください。
[image]

s-katsu, 2021-11-23:
1ヶ月以上更新がないためCloseさせていただきます。
必要ありましたら再Openお願いします。

closed 2021-11-23 by s-katsu