WHILL/mrp-support #24: whill model cr の制御方法について

yoshida5, 2021-11-22:
whill model crを研究に使用させていただいています。
linux ubuntu18.04をインストールしたpcとwhillをシリアル通信で接続し、制御を行いたいのですが、いくつかわからない点があるので質問させてください。
（1）pcとwhillを接続した後、[ WARN] [1637570909.655093825]: Check your serial connection to WHILL.　と表示されるのですが、原因はなんでしょうか。解決策はありますか？
（2）rostopic listの中にcontroller/cmd_velが存在しません。controller/cmd_velを使用するにはどうすればよいでしょうか。
（3）whillをrosで動かす手順を具体的に教えていただけないでしょうか。（前進するだけ等簡単なもので構いません。）

s-katsu, 2021-11-23:
お問い合わせありがとうございます。

まずはこちらのissueを参考にportなどが正しく設定されているかご確認お願いします。

https://github.com/WHILL/Model_CR_Technical_Support/issues/2

1,2に関してはこちらで解決する可能性はあります。

3に関しましてはcmd_velやjoyをpubするnodeを用意してあげれば確認可能です。
rosのtutorialなどみていただければわかるかと思います。
http://wiki.ros.org/ROS/Tutorials/UnderstandingTopics

このように任意のrostopicをコマンドライン上でpublishすることも可能です。

```
$ rostopic pub -1 /turtle1/cmd_vel geometry_msgs/Twist -- '[2.0, 0.0, 0.0]' '[0.0, 0.0, 1.8]'
```

yoshida5, 2021-11-24:
ご返答ありがとうございます。

まず、
`ls -lah /dev/serial/by-id/`
と入力し、
[image]
と出力されたので、ttyUSB0に割り当てられていると確認し、
`echo 'export TTY_WHILL=/dev/ttyUSB0' >> ~/.bashrc`
`source ~/.bashrc`
と入力し、環境変数をTTY_WHILLに変更しました。
その後
`echo $TTY_WHILL`
と入力すると
[image]
と出力され、TTY_WHILLに設定されているポートがttyUSB0となっていることを確認しました。
そして、
`roslaunch ros_whill ros_whill.launch`
と入力しても、
[image]
となってしまい、シリアルポートの接続を確認しろとでます。
その後、
`rostopic list`
と入力すると、
[image]
のように、
cmd_vel等のtopicがでません。
初心者のため、ネットの情報を駆使して色々とやってみたのですが、なにかやり方を間違っているのでしょうか。
長文失礼しました。よろしくお願いします。

s-katsu, 2021-11-29:
すみませんが、こちらのIssueと同じ環境で試されているのでしょうか？（PCのhostnameが同じになっていましたので）
もしケーブル交換でこちらも解決されたようでしたらCloseお願いします。

https://github.com/WHILL/Model_CR_Technical_Support/issues/25#issuecomment-981394417

yoshida5, 2021-11-29:
こちらもケーブルを交換することで解決いたしました。ありがとうございました。