WHILL/mrp-support #25: Whill Model CRの制御方法について

mimimidododo, 2021-11-22:
いまWhill Model CRを用いて研究させていただいています。
使っているｐｃの情報：Linux Ubuntu18.04（ROS　Melodicをインストール済み）
公式サイト（https://github.com/WHILL/ros_whill）の書いた通り色々試しましたが、以下のエラーが出ました。
[image]
「Check your serial connection to WHILL.」の文字が出ましたので、ちゃんとWhillと繋いでいるかどうかを考えています。
また、「rostopic list」というコマンドを実行しました。結果は以下になります。
[image]
公式サイトと比べてcmd_velおよびimu二つトピックは足りませんでした。
解決方法を教えていただきませんか。
どうぞよろしくお願いいたします。

s-katsu, 2021-11-23:
お問い合わせありがとうございます。

> 「Check your serial connection to WHILL.」の文字が出ましたので、ちゃんとWhillと繋いでいるかどうかを考えています。

こちらのissueを参考にportなどが正しく設定されているかご確認お願いします。

#2 

また実際に実行したコマンドを添付いただけるともう少しコメントすることも可能です。

> 公式サイトと比べてcmd_velおよびimu二つトピックは足りませんでした。
cmd_velが表示されないのはconnectionが確立できていないためです。(#2参照）

imuはrobot系の利用に耐えられる精度のものではないため、このcommit以降は無効とさせてもらっています。
お手数おかけして申し訳ありませんが、外付けでIMUをご用意いただくようお願いします。

https://github.com/WHILL/ros_whill/commit/c61f6e8936a8f0dd4db3ffac99603d537b8972c1

mimimidododo, 2021-11-24:
ご返信ありがとうございます。Whill Model CRの制御方法についてもう少し聞かせていただきたいと思います。
「また実際に実行したコマンドを添付いただけるともう少しコメントすることも可能です。」
実際に実行したコマンドおよび使っている装置は以下になります。

1.使用しているノードｐｃ：Ubuntu18.04LTS（ROS　Melodicインストール済み）

2.Whill Model CRと接続用のシリアル変換ケーブルは[こちら](https://www.amazon.co.jp/UGREEN-%E3%82%B7%E3%83%AA%E3%82%A2%E3%83%AB%E5%A4%89%E6%8F%9B%E3%82%B1%E3%83%BC%E3%83%96%E3%83%AB-%E3%82%B7%E3%83%AA%E3%82%A2%E3%83%AB%E3%82%B1%E3%83%BC%E3%83%96%E3%83%AB-D-sub9%E3%83%94%E3%83%B3-Windows/dp/B00QUZY4UG/ref=sr_1_3_sspa?__mk_ja_JP=%E3%82%AB%E3%82%BF%E3%82%AB%E3%83%8A&keywords=ugreen+usb+%E3%82%B7%E3%83%AA%E3%82%A2%E3%83%AB+%E5%A4%89%E6%8F%9B&qid=1637730021&sr=8-3-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUFMTkdYTDc1TkxUSEcmZW5jcnlwdGVkSWQ9QTA1NzQ4ODgzMVRWNlQ2NEM0M1JDJmVuY3J5cHRlZEFkSWQ9QTFMMEhPMDFEQlhHVCZ3aWRnZXROYW1lPXNwX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=)になります。

3.Whill Model CR車体の購入は2021年6月くらいです。（公式サイトに「This command is only available Model CR firmware updated after 2019.12. If you want to use this cmd_vel, please update firmware of Model CR by contact to sales of WHILL.」が書いていますが、「Model CR firmware updated after 2019.12」の条件をあっていると思います。）

4.実際に実行したコマンド：まず、

```
cd ~/catkin_ws/src
git clone https://github.com/WHILL/ros_whill.git
sudo apt install -y ros-melodic-serial
catkin_make
ls -lah /dev/serial/by-id/
```

と入力すると、

[image]

となりました。

次に、以下のコマンドを実行すれば、

```
export TTY_WHILL=/dev/ttyUSB0
source ~/.bashrc
echo $TTY_WHILL
```

[image]

となりました。

最後、

```
roslaunch ros_whill ros_whill.launch
```

を入力し、

[image]

[image]

となってしまいました。

5.ｐｃの問題：

ros_whillをインストールする前に、それぞれUbuntu18.04LTSとUbuntu20.04LTSをインストールしている2台のノートｐｃを用意しました。上記のシリアル変換ケーブルを用いてUbuntu20.04LTSをインストールしているノートｐｃに挿入すれば、[image]

[image]

すぐ「Serial」というファイルが出来、またttyUSB0のファイルも出来ましたが、Ubuntu20.04LTSをインストールしているノートｐｃに挿入すれば反応はありませんでした。そのため、[こちら](https://qiita.com/reo/items/3758b77fd4837b434260)の情報を参考になり、実行した後またシリアル変換ケーブルを挿入し、Ubuntu20.04LTSと同じく「Serial」ファイルが出来ました。この上で、手順4のコマンドを実行し、最後のエラーになってしまいました。

インターネットに色々探しましたが、よく解決策を見つかりませんでした。初心者ですので、よろしければ、解決策を詳しく教えていただければと思います。

mimimidododo, 2021-11-29:
すいません、新しいシリアル変換ケーブルを購入しました。新しいケーブルで試して制御出来ました。この前の返信ありがとうございます。

closed 2021-11-29 by mimimidododo