WHILL/mrp-support #26: Noeticでのserial接続について

SAR-abe, 2021-11-24:
raspberry pi 3B+(Ubuntu20.04 server LTS 32bit ,noetic)でのwhill開発を検討しています．
現在，検討環境にてros_whillパッケージがビルドできない状況です．

事前確認としてUbuntu18.04 server LTS 32bit ,melodic 環境にて動作確認済みです．
その際，ビルド時に”serial”パッケージが必要とのことで以下のコマンドからインストールを行いました．
sudo apt install  ros-melodic-serial

しかし，noetic環境では以下のコマンドからserialパッケージをインストールできませんでした．
sudo apt install ros-noetic-serial
E: パッケージ ros-noetic-serial が見つかりません

noeticでは従来のserialパッケージをサポートしていないようです．解決策がございましたらご教示いただけると幸いです．

s-katsu, 2021-11-26:
お問い合わせありがとうございます。

これはどちらかというとROSのserialがnoeticで使えるかという話になりますので、弊社製品に関する内容ではないため、お答えできないのです。

ただ元のソースのrepository のissueでやりとりではソースコードをビルドすれば使えるという話が出ているようです。

https://github.com/wjwwood/serial/issues/222#issuecomment-688124660

SAR-abe, 2021-11-26:
ご回答ありがとうございます．

提示していただいたissueをもとにserialをソースコードからビルドして,ros_whillパッケージをビルドすることができました．

対象外の質問にもかかわらず，親切に対応して頂きありがとうございました．

closed 2021-11-26 by SAR-abe