WHILL/mrp-support #28: ファームウェアのアップデートに関して

MasashiHirai1993, 2022-01-19:
Model CRのファームウェアのアップデート方法を教えていただくことは可能でしょうか？
また、アップデートを遠隔で行うことは可能でしょうか？

現状の最新版のファームウェアは、以下のものが最新との認識ですが、その他の最新のアップデート情報などありますでしょうか？
https://github.com/WHILL/ros_whill

お手数ですが、上記ご教示いただけると幸いです。

s-katsu, 2022-01-19:
お問い合わせありがとうございます。
CRのファームウェアのアップデートはWHILLの営業が窓口となっておりますので、購入時にやりとりされていました営業がおりましたらそちらへメールでお問い合わせください。
もしくはこちらからお問い合わせください。
https://whill.inc/jp/whill-contact#content-whill-private-customer

具体的なアップデート方法自体は営業から連絡がいくかたちとなります。

> 現状の最新版のファームウェアは、以下のものが最新との認識ですが、その他の最新のアップデート情報などありますでしょうか？
https://github.com/WHILL/ros_whill

こちらはrROS1のライブラリになりますので、機体の中のファームウェアとは別のものです。
ただし、ros_whillの最新はCRの最新ファームウェアで動作保証するものとなっております。

s-katsu, 2022-01-20:
@MasashiHirai1993 こちら解決済みでしたらCloseお願い致します。

MasashiHirai1993, 2022-01-20:
ご回答ありがとうございました。
営業の方にご対応いただきました。

closed 2022-01-20 by MasashiHirai1993