WHILL/mrp-support #29: teleop.pyの使い方について

mimimidododo, 2022-02-04:
いまWhill Model CRを用いて研究させていただいています。
一先ず、キーボードでWHILLを制御しようと思いますが、端末にroscoreが起動する状態でrosrun ros_whill teleop.pyというコマンドを入れ、結果は以下となりました。
[image]
whill.msgがなければ、実行できなさそうです。rosmsg listを実行し、確かwhill.msgが有りません。
[image]
解決方法を教えていただきませんか。
どうぞよろしくお願いいたします。

s-katsu, 2022-02-24:
ご連絡ありがとうございます。
お返事大変遅れて申し訳ありません。

こちらのteleop_whill.pyは以前の古いコードに対応したもので、今はサポートしていませんので削除致します。

ros_whillのREADMEに記載されておりますが、通常のsensor_msgs/Joyでやりとりできますので、
適宜キーボードかジョイスティックでtopicをpubしていただければ動かすことは可能です。

https://github.com/WHILL/ros_whill#controllerjoy-sensor_msgsjoy

ネットで拾ったレベルですが参考情報です。
https://qiita.com/srs/items/9114bb3c27a148e0b855

例えばjoy_nodeを使う場合はこのようなlaunchをかけばジョイスティックで操作できるはずです。
```
<launch>
    <node name="joynode" pkg="joy" type="joy_node" args="" output="screen" required="true">
        <param name="autorepeat_rate" value="100"/>
        <remap from="/joy" to="/whill/controller/joy" />
        <param name="dev" value="/dev/input/js0"/>
    </node>  
</launch>
```

mimimidododo, 2022-02-24:
ありがとうございます。参考させていただきます。

closed 2022-02-24 by mimimidododo