WHILL/mrp-support #30: ArduinoからのWHILL Model CR制御について

mSHiOm, 2022-04-08:
ArduinoでのWHILL Model CR制御を試みているのですが，WHILLとの通信が出来ません．

[2018アドベントカレンダーの記事](https://qiita.com/aquahika/items/14a129db401926485bda)　を参考に
while-arduino-sdkの中のShakeサンプルをArduinoに書き込んだところ，
Arduinoボード上のRX, TX LEDは光りますが，Whillが動きません．
その他，Callbackのサンプルも試しましたが，Serialへのプリントが出ない状態です．

Arduino制御を始めるにあたり，
WHILLの起動方法や電源供給などの注意点はありますでしょうか．

ご回答よろしくお願いいたします．

使用機材：
- Arduino Uno R3
- RS232Cシールド
- ModelCR to RS232Cケーブル（Whillから購入したもの）

以上を下の図のように繋いでいます．
[image]

mSHiOm, 2022-04-11:
[追記]　RS232Cの連結部分に[RS232C NULLモデム](https://www.amazon.co.jp/dp/B018AHEDB6/ref=cm_sw_r_cp_ep_dp_Q3rbCbNAWRN5R)を使用しても同じ症状です．

s-katsu, 2022-04-18:
@mSHiOm 返信遅れて大変申し訳ございません。
こちらで同じ状況確認してみましたが、
https://github.com/WHILL/whill-sdk-arduino
と

- Arduino Uno
- RS232Cシールド
- RS232Cモデム

の組み合わせで問題なく通信できました。

RS232Cモデムとは機体に接続するDsubのケーブルコネクタを直接接続しました。

接続の様子も添付します。
RS232Cシールド上はTX: 7pin, RX: 6pinとしています。（画像参照）
[image]
[image]
[image]

exampleのblink_power.inoを若干書き換えた下記のinoファイルを実行しました。(実はpower onは2回連続してコマンドを送信する必要があるためその点だけ変更しています。）

```
#include <SoftwareSerial.h>
#include "WHILL.h"

SoftwareSerial ss(7,6);
WHILL whill(&ss);

void callback_powered_on(WHILL* caller)
{
  // This function is called when powered on via setPower() 
  Serial.print("powered_on");
}

void setup() {
  // put your setup code here, to run once:

  Serial.begin(9600);

  //Register callbacks
  whill.register_callback(callback_powered_on,WHILL::EVENT::CALLBACK_POWER_ON);

  whill.begin(50);   // Set refresh interval of WHILL
}

void loop() {
  // put your main code here, to run repeatedly:
  whill.refresh();

  Serial.println("power on");
  whill.setPower(true);  // Turn WHILL on
  whill.delay(100);
  whill.setPower(true);  // Turn WHILL on
  whill.delay(5000);

  Serial.println("power off");
  whill.setPower(false);  // Turn WHILL off
  whill.delay(5000);  
}
```

お手数ですが再度接続状況などご確認いただけますでしょうか。
もしくはpythonなどのライブラリもありますのでそちらも合わせてご確認いただけると問題の切り分けになるかと思います。

https://github.com/WHILL/pywhill

mSHiOm, 2022-04-19:
ご返事ありがとうございます．

上記のプログラムでblink_power.inoを試してみましたが，Whillの反応がありませんでした．
（シリアルモニタではpower on/off が出るのでArduino-PCの通信問題ではないと思います．）

また，PyWhillの方も試してみましたが，
[cr_example3_callback.py](https://github.com/WHILL/pywhill/blob/master/example/cr_example3_callback.py)を実行してもWhillからの反応はありませんでした．

現在の接続は以下のようになっています．
[image]

s-katsu, 2022-04-19:
機体の左アームの電源ボタンで電源を入れた状態で
https://github.com/WHILL/pywhill/blob/master/example/cr_example4_blink_power.py
を実行しても電源切れない状態でしょうか？

pywhillは
```
    whill = ComWHILL(port='/dev/ttyUSB0')
```
ポートの設定を正しくしていただかないと接続できないですがそのあたりは問題ないでしょうか？
RS232C経由のUSBシリアルをPCに接続して追加されたポートであれば問題ないです。

もし、cr_example4_blink_power.pyでも電源オフされないとなりますが、故障している可能性があります。
その際はお手数ですがこちらの修理についてというページからご連絡お願い致します。
https://whill.inc/jp/support/repair

mSHiOm, 2022-04-20:
cr_example4_blink_power.pyを実行しましたが，電源は切れませんでした．

今回は，RS232-USBコネクタの都合でwindowsを使用しました．
port = 'COM#' ( # には特定の数字)を入れて実行しましたが，
エラーは出ませんが，power_on_callback()の出力も表示されない状態です．

修理ページから連絡をいたします．ご対応ありがとうございました．

s-katsu, 2022-04-20:
> 修理ページから連絡をいたします

はい、お手数おかけしますがその対応でお願いいたします。

こちらはCloseさせていただきますが、必要あればReopenしてお問い合わせください。
お問い合わせありがとうございます。

closed 2022-04-20 by s-katsu