WHILL/mrp-support #31: xacroの起動について

mayo233, 2022-04-28:
現在，whillをgazeboで動かしたいと考えています．そのため，$source devel/setup.bashをして
 $roslaunch ros_whill model_whill.launchとコマンドを実行した結果，下記のようにターミナル上に表示されてgazeboを起動しても何も表示されません．

suzukamayo@suzukamayo-N850EP6:~/ros_ws/lidar/whill_ws$ roslaunch ros_whill model_whill.launch
... logging to /home/suzukamayo/.ros/log/15f3658e-c6d5-11ec-8437-c0b6f9c7d913/roslaunch-suzukamayo-N850EP6-8117.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/suzukamayo/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.
started roslaunch server http://suzukamayo-n850ep6:33297/
SUMMARY
========
PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.11
NODES
ROS_MASTER_URI=http://localhost:11311/
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete

どのような手順を踏めばgazebo上にwhillを表示させて動かすことが出来ますか？

s-katsu, 2022-05-12:
返答遅れて申し訳ありません。

ros_whillはModel CRのROS用のドライバ packageでしかなく、gazeboのシミュレーションを想定したものではありません。

Model CRとのシリアル通信をROS topic の pub/subに変換するpackageになります。

Gazebo上でロボットをシミュレーションで動かしたい場合は一般的なROSの書籍やWeb Siteで情報ご確認ください。

s-katsu, 2022-05-31:
こちらお返事無いようですのでCloseさせていただきます。
何かありましたらまた再Openしてください。

closed 2022-05-31 by s-katsu