WHILL/mrp-support #33: whillのpythonでの操作時の速度について

tsukamm, 2022-05-12:
私はWHILL model CRをpythonで操作したいと考えています。
その際　whill.send_joystick(int(30), int(0))　のようにintの後ろの数字を変えて速度を調整していました。
しかし以前は127の値で時速およそ6kmほど出ていたのですが、現在時速2kmほどしか出なくなってしまいました。
どのような問題があると考えられるでしょうか？
ご教示お願い致します。

s-katsu, 2022-05-20:
お問い合わせありがとうございます。

> しかし以前は127の値で時速およそ6kmほど出ていたのですが、現在時速2kmほどしか出なくなってしまいました。
Joystickコマンドの設定値は整数値で%を表していますので、±100が最大値で、機体のJoystickを最も倒した場合と同等の操作量を想定しています。

機体のアーム先端についている速度切り替えボタンはCRのJoystickコマンドの速度とは連動しておらず、CRの通信用のspeed profileを持っています

こちらが設定変更されている場合は速度が遅くなっている可能性があります。

speed profileの設定についてはこちらのExampleを参考にしてください。
https://github.com/WHILL/pywhill/blob/master/example/cr_example6_copy_speed_profile.py

speed profileの仕組み自体はこちらの通信仕様書の3.2.5 SetSpeedProfileなどを参照ください。
https://github.com/WHILL/whill_control_system_protocol_specification/blob/master/WHILL_Control_System_Protocl_Specification.pdf

基本的にspeed profileは通信仕様書にある仕組みをpythonで利用できるようにしたものになります。

[image]

また、CRの通信ケーブルを使用せずに、そもそも機体のJoystickを使った通常の走行でも速度が遅い場合は別の要因の可能性があります。

tsukamm, 2022-05-20:
speed profile の設定変更によって解決いたしました。
誠にありがとうございました。

closed 2022-05-20 by tsukamm