WHILL/mrp-support #35: ArduinoとWHILLとcallback

unkasu1987, 2022-07-27:
WhillをArduinoで制御するために御社が公開されている「whill-sdk-arduino-master.zip」のプログラムをArduinoに書き込むため、「WHILL.h」と「callback」をArduinoIDEで読み込み、書き込みましたがシリアルモニターにジョイスティックの入力値が表示されませんでした。またArduino IDE内でシリアルポートを選択することもできませんでした。解決策をお願いいたします。
[image]

s-katsu, 2022-08-04:
お問い合わせありがとうございます。

>「WHILL.h」と「callback」をArduinoIDEで読み込み、書き込みましたがシリアルモニターにジョイスティックの入力値が表示されませんでした。

こちらはArduinoのSampleのCallbackを動かしてみても動作しなかったということでしょうか？
また、サンプルのblink_powerなど動かしても同様に電源入らない状況でしょうか？

サンプルに使用方法はこちらご参照ください。

https://qiita.com/aquahika/items/14a129db401926485bda#%E3%82%B5%E3%83%B3%E3%83%97%E3%83%AB%E3%82%92%E5%8B%95%E3%81%8B%E3%81%97%E3%81%A6%E3%81%BF%E3%82%88%E3%81%86

> またArduino IDE内でシリアルポートを選択することもできませんでした。

こちらはArduino IDEのシリアルポートというのはIDEのUI上の
ツール/シリアルポート
のことをおっしゃっているのでしょうか？（下図参照）

[image]

また以前同じようなお問い合わせがありましたが(下記のリンク参照）、その際は
RS232Cシールド上のジャンパーピンとRX/TXのポートをご確認いただいております。
こちらも併せてご確認お願い致します。
https://github.com/WHILL/Model_CR_Technical_Support/issues/30#issuecomment-1101321700

unkasu1987, 2022-08-04:
シリアルポ`トはxkできるようになりましたが、依然callbackやmirror_joystickは婴护蓼护螭扦筏俊％弗悭螗雪`ピンはTX7とRX6につないでいます。Arduinoへの源の供oも行っています。

Windows の メ`ル<https://go.microsoft.com/fwlink/?LinkId=550986> から送信

差出人: Shunsuke ***@***.***>
送信日r: 2022年8月4日 18:28
宛先: ***@***.***>
CC: ***@***.***>; ***@***.***>
件名: Re: [WHILL/Model_CR_Technical_Support] ArduinoとWHILLとcallback (Issue #35)

おい合わせありがとうございます。

「WHILL.h」と「callback」をArduinoIDEでiみzみ、zみましたがシリアルモニタ`にジョイスティックの入力が表示されませんでした。

こちらはArduinoのSampleのCallbackを婴筏皮撙皮幼鳏筏胜盲郡趣いΔ长趣扦筏绀Δ
また、サンプルのblink_powerなど婴筏皮馔に源入らない状rでしょうか？

サンプルに使用方法はこちらご参照ください。

https://qiita.com/aquahika/items/14a129db401926485bda#%E3%82%B5%E3%83%B3%E3%83%97%E3%83%AB%E3%82%92%E5%8B%95%E3%81%8B%E3%81%97%E3%81%A6%E3%81%BF%E3%82%88%E3%81%86

またArduino IDE内でシリアルポ`トをxkすることもできませんでした。

こちらはArduino IDEのシリアルポ`トというのはIDEのUI上の
ツ`ル/シリアルポ`ト
のことをおっしゃっているのでしょうか？（下聿握眨

[image]<https://user-images.githubusercontent.com/20053970/182812799-a31da1af-5952-431e-8519-e3a5400048f4.png>

また以前同じようなおい合わせがありましたが(下のリンク参照）、そのHは
RS232Cシ`ルド上のジャンパ`ピンとRX/TXのポ`トをご_Jいただいております。
こちらも悚护皮创_Jおい致します。
#30 (comment)<https://github.com/WHILL/Model_CR_Technical_Support/issues/30#issuecomment-1101321700>

s-katsu, 2022-08-05:
すみません、文字化けしていますので再度お送りいただけますか？

unkasu1987, 2022-08-05:
シリアルポートは選択できるようになりましたが、依然callbackやmirror_joystickは動かせませんでした。ジャンパーピンはTX7とRX6につないでいます。Arduinoへの電源の供給も行っています。

iPhoneから送信

s-katsu, 2022-08-08:
ご回答ありがとうございます。

https://github.com/WHILL/Model_CR_Technical_Support/issues/30#issuecomment-1101321700

上記のリンクに記載している

> exampleのblink_power.inoを若干書き換えた下記のinoファイルを実行

を試していただけますか？

また、pythonなどのライブラリもありますのでそちらも合わせてご確認いただけますか？問題の切り分けになるかと思います。

https://github.com/WHILL/pywhill

unkasu1987, 2022-08-25:
ありがとうございます、inoファイルを実行とはどうすればいいでしょうか。Arduinoに書き込むのでしょうか。それともパソコンとarduinoを接続して実行するのでしょうか？

iPhoneから送信

ご回答ありがとうございます。

#30 (comment)<https://github.com/WHILL/Model_CR_Technical_Support/issues/30#issuecomment-1101321700>

https://github.com/WHILL/pywhill

unkasu1987, 2022-08-31:
「blink_powerを若干Qえた下のinoファイル」をg行しましたが源がオフになったりはしませんでした。Pythonのライブラリは使い方がわかりませんでした。

Windows の メ`ル<https://go.microsoft.com/fwlink/?LinkId=550986> から送信

差出人: Shunsuke ***@***.***>
送信日r: 2022年8月8日 12:37
宛先: ***@***.***>
CC: ***@***.***>; ***@***.***>
件名: Re: [WHILL/Model_CR_Technical_Support] ArduinoとWHILLとcallback (Issue #35)

ご回答ありがとうございます。

#30 (comment)<https://github.com/WHILL/Model_CR_Technical_Support/issues/30#issuecomment-1101321700>

上のリンクにdしている

exampleのblink_power.inoを若干Qえた下のinoファイルをg行

をしていただけますか？

また、pythonなどのライブラリもありますのでそちらも合わせてご_Jいただけますか？}の切り分けになるかと思います。

https://github.com/WHILL/pywhill

s-katsu, 2022-08-31:
> ありがとうございます、inoファイルを実行とはどうすればいいでしょうか。Arduinoに書き込むのでしょうか。それともパソコンとarduinoを接続して実行するのでしょうか？

申し訳ありませんが、まずはArduinoの一般的な使い方をご理解いただいたうえでお問い合わせいただけますでしょうか。
ArduinoやM5Stackなどで、簡単なプログラム（LEDを点滅させるなど）を動作させることがご理解いただけたうえで弊社の製品の通信についてご確認させていただきたいです。

>Pythonのライブラリは使い方がわかりませんでした。
Pythonのライブラリは使い方はGithubリポジトリのREADMEに記載されております。

https://github.com/WHILL/pywhill#pywhill

unkasu1987, 2022-08-31:
ちゃんとメ`ル全部てください。Inoファイルをg行しても婴蓼护

Windows の メ`ル<https://go.microsoft.com/fwlink/?LinkId=550986> から送信

差出人: Shunsuke ***@***.***>
送信日r: 2022年8月31日 13:17
宛先: ***@***.***>
CC: ***@***.***>; ***@***.***>
件名: Re: [WHILL/Model_CR_Technical_Support] ArduinoとWHILLとcallback (Issue #35)

ありがとうございます、inoファイルをg行とはどうすればいいでしょうか。Arduinoにzむのでしょうか。それともパソコンとarduinoを接Aしてg行するのでしょうか？

申しUありませんが、まずはArduinoの一般的な使い方をご理解いただいたうえでおい合わせいただけますでしょうか。
ArduinoやM5Stackなどで、gなプログラム（LEDを点绀丹护毪胜桑─幼鳏丹护毪长趣蠢斫猡い郡坤堡郡Δà潜咨绀窝u品の通信についてご_Jさせていただきたいです。

Pythonのライブラリは使い方がわかりませんでした。
Pythonのライブラリは使い方はGithubリポジトリのREADMEにdされております。

https://github.com/WHILL/pywhill#pywhill

s-katsu, 2022-09-20:
本件その後いかがでしょうか。問題解消しましたらIssueのCloseをお願いします。
GitHub Issue上でCloseを押してください。

closed 2022-09-20 by unkasu1987