WHILL/mrp-support #36: ModelCRにてシリアル通信ができない状況となっておりアドバイスがほしいです。

KM-tajima, 2022-07-29:
■状況説明
・https://github.com/WHILL/pywhillのサンプルコードで疎通確認実施中だが筐体側は無反応という状況です
・以前某所にてModelCR(本件とは異なる筐体)にて実験した際、上記コードで問題無く制御可能であった。
　(callbackで値が取れたり、スピードモードの変更、電源のONOFFなどの制御ができた) 
・現在実験に使用している筐体は、上記実験時に使用したものと全く同じパソコンを使用し、同じ環境(pythonやwhillのコードのバージョンやケーブル類)で疎通確認しているが無反応。
・現状疎通確認実施しているModeCRはシリアルNoは「C_2005120_JP」です
・コネクタケーブルはwhillより販売されている「https://whill.inc/jp/news/22675」を利用しております。
・コネクタケーブルのヒューズが飛んでいる形跡はなし。
・手動でのJoystickの運転操作は問題無くできている。
・下記サンプルコードにてresponseをprintすると「5」が返るため、USB/232c変換ケーブルへのCOMポートへのCOMMANDのwrite処理は成功している。(しかし筐体側は無反応)

〇cr_example4_blink_power.py
```
def main():
    whill = ComWHILL(port='COM3')
    whill.register_callback('power_on', power_on_callback)
    print("start ")
    while True:
        whill.sleep(3)
        response = whill.send_power_off()
        print(response)

        whill.sleep(3)
        response = whill.send_power_on()
        print(response)
```
■相談内容
・どのあたりを確認したらいいかなどアドバイス頂けないでしょうか？
・コネクタケーブルまわりで通信不良が無いかの確認手段などあれば情報頂けないでしょうか？
・または、そちらに筐体を送付し見て頂く必要あるでしょうか？

s-katsu, 2022-08-01:
お問い合わせありがとうございます。
状況確認中です。追ってご回答させていただきますので、もう少々お待ちください。

s-katsu, 2022-08-02:
本件ですが、納品されたModel CRの機体のファームウェアに問題があったため、アップデートさせていただきます。
ご迷惑をおかけして申し訳ありません。

営業からご連絡いくと思いますので恐れ入りますがもう少々お待ちください。

もしご連絡なければこちらからアップデートの旨お問い合わせお願い致します。
https://whill.inc/jp/whill-contact#content-form-corporate-customer-rental

s-katsu, 2022-09-05:
こちらCloseさせていただきます。
また問題があればRepoenや別Issueの起票お願いします。

closed 2022-09-05 by s-katsu