WHILL/mrp-support #37: WHILL Model CRのROS制御について

jonppy, 2022-08-01:
研究でWHILL Model CRを使っています。先日、新たにModel CRをもう一台購入しました。ROSを用いてWHILLの制御を行っているのですが、今まで使用してきたWHILLはROSのトピックのwhill/controller/joyで制御しており、新しいWHILLではwhill/controller/cmd_velのトピックで制御できるのでしょうか。また、ファームウェアをアップデートすることで以前のWHILLでもwhill/controller/cmd_velを用いることができるのでしょうか。アップデートが可能であるのならば至急行いたいのですが、どのような手順で行えばよろしいでしょうか。お忙しいところ恐縮ですがご返信お待ちしております。

s-katsu, 2022-08-02:
お問い合わせありがとうございます。

> また、ファームウェアをアップデートすることで以前のWHILLでもwhill/controller/cmd_velを用いることができるのでしょうか。アップデートが可能であるのならば至急行いたいのですが、どのような手順で行えばよろしいでしょうか。お忙しいところ恐縮ですがご返信お待ちしております。

`/whill/controller/cmd_vel`はアップデート済みの機体であれば使用可能です。
新しいWHILLがいつ頃購入されたかにもよりますが、この1年以内に購入したものであればcmd_vel対応されていると推測されます。
古い機体もアップデートを行えば`/whill/controller/cmd_vel`対応となります。

> アップデートが可能であるのならば至急行いたいのですが、どのような手順で行えばよろしいでしょうか。

アップデートは購入した際の営業にご連絡いただくか、こちらからお問い合わせください。
https://whill.inc/jp/whill-contact#content-form-corporate-customer-rental

jonppy, 2022-08-03:
ご返信ありがとうございます。

新しいWHILLは先月(2022年7月)に購入したものですが，ファームウェアは最新バージョンなのでしょうか．また，そのバージョンの確認方法はありますか．

s-katsu, 2022-08-04:
> 新しいWHILLは先月(2022年7月)に購入したものですが，ファームウェアは最新バージョンなのでしょうか．また，そのバージョンの確認方法はありますか．

機体番号を教えていただければ確認して結果をお伝えいたします。

もしこちらのgithub issueがOpenですのでここに機体番号を書きにくい場合は、こちらからお問い合わせください。
https://whill.inc/jp/whill-contact#content-form-corporate-customer-rental

jonppy, 2022-08-04:
ご返信ありがとうございます．そちらのリンクから問い合わせてみます．
このissueはcloseします．

closed 2022-08-04 by jonppy