WHILL/mrp-support #38: 緊急停止機能について

imossan, 2022-09-02:
別の方が質問されていますが、それとは少し違うため質問させていただきます。
別スレッドでserial通信を遮断すると書いてありましたが、その方法では直前の速度指令による影響を受ける可能性があり、安全では無いと考えています。
バッテリーから出る配線を直接遮断させることで停止させたいと考えておりますが、これについては可能でしょうか。
難しいようであれば、他に緊急停止スイッチを付け加える方法は無いでしょうか。

また、/whill/controller/cmd_velを受信し動いている際に、whill本体に付いているjoyスティックでマニュアルでオーバーライドさせることができれば緊急停止スイッチは必要ないと考えています。
（現在はオーバーライドができない）

KazumichiShirai, 2022-09-05:
> バッテリーから出る配線を直接遮断させることで停止させたいと考えておりますが、これについては可能でしょうか。

ケーブルを用意していただければ可能ですが、コネクタが特殊ですので、現実的ではないと思います。
> 他に緊急停止スイッチを付け加える方法は無いでしょうか。

申し訳ございません。すぐには思い付かないです。

> 現在はオーバーライドさせることはできませんが、これを実現することはできますか？

実現できません。

他の方法として、Model CRの手元の電源ボタンを押していただく方法があります。
手元の電源ボタンを押していただくと、強制的に機体は電源OFFになります。
この方法はいかがでしょうか？

imossan, 2022-09-06:
ご返信ありがとうございます。

緊急時の場合、なるべく直感的に停止の動作を行いたいと考えており、新しくボタンを配置したいと考えていました。
他の方法は電源ボタンを押すしかないとのことですので、操作ボックスの中にある、プラスの配線を緊急停止スイッチに繋げることで実現しようと思います。
ありがとうございました。

KazumichiShirai, 2022-09-06:
ご検討ありがとうございます。お手数ですが、よろしくお願いします。
念のため、配線を繋げる前に、電源ボタンを押して想定の挙動になるかは事前にご確認していただけると安心です。

問題なさそうでしたら、本件はcloseしていただけると幸いです。

closed 2022-09-06 by imossan