WHILL/mrp-support #39: Whillの速度制御について

ninokin0319, 2022-09-19:
WHILL技術者様

WHILLの相談窓口に以下のような内容で連絡させていただいたところ，こちらを紹介されましたので，こちらに質問させていただきます．
他の方に，質問が共有されるようですので，個人情報に関する部分は，除外しています．

―――以下質問―――
WHILLについて伺いたいことがあり連絡いたしました．
大学の研究室での実験で2台のWHILLをすれ違わせるような課題を行う予定なのですが，2台の速度が異なることに気が付きました．
実際，直線のコースを競争させると，1台のほうが明らかに速度が遅いことが確認されました．

積載量の違いである可能性も考え，早いほうのWHILLに重い荷物を積んで，遅いほうのWHILLを軽くしても問題は解決しませんでした．

つきましては，以下の点を伺いたく思います．
速度差が生じる原因としてどのようなことが考えられるのか？
それらを修理(修正？)することは可能か？

ちなみにそれぞれの製品番号は以下の通りです．
遅いほうのWHILL：C_2004223_JP
早いほうのWHILL：C_1806268_JP

お忙しいところ恐縮ですが，どうぞご教授いただければ幸いです．

KazumichiShirai, 2022-09-19:
ご連絡ありがとうございます。
RS232C経由で制御したときに、2台のWHILL Model CRに速度差が生じているという理解で正しいでしょうか？ (WHILL Model CRのjoystickを操作した場合ではないと考えております。)
RS232C経由で制御している前提で回答させていただきます。

考えられる原因を列挙させていただきます。
1. バッテリー残量の違い
バッテリーを交換しても同じ結果でしょうか？
使用されている直線の斜度によっては、バッテリーが満充電付近の場合、速度を制限することがあります。
また、バッテリーの残量が少ない場合、速度が制限されることがあります。 積載重量や斜度によるのですが、15%あれば差は生じないと思います。

2. 設定の違い
すでにご確認済みかもしれませんが、RS232Cで制御したときの最高速度設定に差分はないでしょうか？
確認方法は、
StartSendingDataで、D0=0, S0=4にすればspeed profileを確認できます。
WHILL Control System Protocol Specificationの3.2.1と4.2.1をご確認ください。

また、お手数ですが、以下の動作確認や状況のご回答いただけないでしょうか？
1. 速度を比較したときに使ったコマンドの詳細を教えていただけないでしょうか？ SetJoystickでしょうか？SetVelocityでしょうか？またコマンドの中身を教えていただけると幸いです。
2. WHLL Model CRのjoysitckを使って、速度を比較しても同様に速度差が生じているか確認していただけないでしょうか？

ninokin0319, 2022-09-20:
丁寧なご対応ありがとうございます．
先に申し上げておきますと，私は技術者ではありませんので，技術的な点については私だけでは判断が付きかねるという点だけ，予めご了承ください．

まず，最初の質問ですがむしろ
”WHILL Model CRのjoystickを操作した”状況での現象です．

といいますのも，今回の実験では，実際に人間を載せてその際のインタラクションについて検討しますので，人間が操作したとき…という前提があります．
なので今回速度が違った状況というのは，二人の人間がジョイスティックを思いっきり前に倒したときの，速度差になります．

まず，バッテリーの違いについてですがどちらもフル充電で実施いたしました．
準備等で，多少は減っていましたが80%を下回るようなことはありませんでした．

2点目についてですが，WHILLの操作に詳しい方に伺ってみたいと思います．
伺うにあたって，以下の点について確認させてください．
１．この設定はジョイスティックで操作する際にも影響するものでしょうか？

最後の２点について回答いたします．
A1.コマンドは使用しておりません．
A2.まさにそれが現在生じている問題です．速度差を確認しました．

以上，専門知識がないので的を射た回答になっていない部分があるかもしれませんが，どうぞよろしくお願いいたします．

KazumichiShirai, 2022-09-23:
ご返信ありがとうございます。
状況、より理解できました。

> １．この設定はジョイスティックで操作する際にも影響するものでしょうか？

StartSendingData commandではジョイスティック操作の速度に影響は与えません。
しかし、SetSpeedProfile commandでジョイスティック操作の速度設定値を変更することができます。このcommandで速度設定値を変更していれば、ジョイスティック操作の速度に影響を与えます。

また、WHILL model CR(及び、WHILL model C)は専用のスマートフォンアプリで、モードを変更することで、速度設定値を変更することができます。もし可能なら、このアプリを使って、両方の機体を同じモードにしてみてください。
スマートフォンアプリ操作ガイドはこちらになります。
https://whill.inc/jp/wp-content/themes/whill-jp/pdf/whill_model_c_app_guide.pdf

ninokin0319, 2022-09-27:
ご回答ありがとうございます．
両機体の間に設定の違いがあったようです．

まだテストはしておりませんが，おそらくこれで問題が解決したと思われます．

もし，次回走行テストで速度差があるようであればまた相談させていただきます．

この度は迅速なご対応ありがとうございます．

KazumichiShirai, 2022-09-27:
ご返信ありがとうございます。
速度差について、ご確認いただき、問題が解決されていそうであれば、本件のcloseをお願いします。
引き続き、問題がありそうでしたら、ここに状況を記載していただけたらと思います。

ninokin0319, 2022-10-19:
遅くなりましたが，確認ができました．
特に問題はありませんでしたので，本件をcloseさせていただきます．

どうもありがとうございました．

closed 2022-10-19 by ninokin0319