WHILL/mrp-support #40: WHILL Model CRの仕様について質問

keita1523, 2022-09-30:
ROS2を使って制御できる車両を探しており、貴社のWHILLの購入を検討しております。
いくつか質問させてください。
（同じ質問をお問い合わせ窓口に送ったところ、こちらを紹介されました）

１．制御の誤差などはあるか
　　　車両の制御誤差はどの程度でしょうか。
　　　（信号を送ってからXミリ秒後に反映される等）

２．制御方法は左右のタイヤの回転速度を直接送信するのか、
　　それとも前後速度コマンドと左右回転コマンドを送るのか

３．手動で動かしている状態から自動モードへの変更はできるか
　　　手動モードで運転している状態で、前方に人が現れたら停止、
　　　といった実験を行いたいと思っています。
　　　そのため手動モードの状態でも、弊社システムから停止信号を送れるのかが気になります。

４．LiDARはついてくるのか。LiDARを取り付ける棒は付属品か。
　　　（Model CRのページにLiDARが取り付けられているイメージ写真があったため）
　　　（もしYesならば）LiDARはVLP16か？

５．ROS対応はMelodicまで、という認識であっているか。
　　　弊社が受けた依頼はROS2で動かしたいという要望でした。
　　　もしROS2に対応しているのであればうれしいですが、
　　　対応してなくてもROS2-ROS1ブリッジなどで
　　　実現可能だと思っています。

s-katsu, 2022-10-03:
お問い合わせありがとうございます。

> １．制御の誤差などはあるか
> 車両の制御誤差はどの程度でしょうか。（信号を送ってからXミリ秒後に反映される等）

恐れ入りますが、こちらは仕様として公開はしておりません。
通信仕様(Baud Rateなど）に関してはこちらの仕様書をご参照ください。

https://github.com/WHILL/whill_control_system_protocol_specification/blob/master/WHILL_Control_System_Protocl_Specification.pdf

> ２．制御方法は左右のタイヤの回転速度を直接送信するのか、
> 
> それとも前後速度コマンドと左右回転コマンドを送るのか

これはライブラリによって異なりますが、ROS1の場合は前進・旋回の速度指令値を送るコマンドと、Joystick相当の入力値を送るコマンドがあります。
具体的にはmsgとしてsensor_msgs/Joyとgeometry_msgs/Twistになります。下記リンク先をご参照ください。

[https://github.com/WHILL/ros_whill#subscribed-topics](https://github.com/WHILL/ros_whill#subscribed-topics)

> ３．手動で動かしている状態から自動モードへの変更はできるか　　　
> 
> 手動モードで運転している状態で、前方に人が現れたら停止、といった実験を行いたいと思っています。　　　そのため手動モードの状態でも、弊社システムから停止信号を送れるのかが気になります。

手動モードというのは、機体についているジョイスティックでの操作だと考えられますが、その前提で回答させていただきます。（もし間違っていればご指摘ください。）

機体のジョイスティックで操作している際に、人が現れたら停止するように外部から制御することは可能です。

例えば下記のようなかたちになります。

- 人が前にいないときは、機体のジョイスティック操作量をそのまま外部コンピュータからジョイスティック操作量として設定
- 人が前にいるときは、外部コンピュータからのジョイスティック操作量を0として設定

> ４．LiDARはついてくるのか。LiDARを取り付ける棒は付属品か。
> 
> （Model CRのページにLiDARが取り付けられているイメージ写真があったため）
> 
> （もしYesならば）LiDARはVLP16か？

LiDARは付属しておりません。

LiDARを取り付けつ棒も付属しておりません。

Model CRは機体本体に通信用のRS232Cケーブルが付いているモデルとなります。

> ５．ROS対応はMelodicまで、という認識であっているか。
> 
> 弊社が受けた依頼はROS2で動かしたいという要望でした。
> 
> もしROS2に対応しているのであればうれしいですが、対応してなくてもROS2-ROS1ブリッジなどで実現可能だと思っています。

ROS2はWHILL社では正式にはサポートしておりません。

ただし、社外含めた有志のメンバーでのGitHub Group : whilllabの方にROS2のライブラリは存在しています。ただし、crystalまでしかメンテナンスされていないです。

https://github.com/whill-labs/ros2_whill

keita1523, 2022-10-15:
ご返答ありがとうございます。
＞手動モードというのは、機体についているジョイスティックでの操作だと考えられますが、その前提で回答させていただきます。（もし間違っていればご指摘ください。）
はい。その認識です。
できるとのこと承知しました。

追加で3点質問させてください。
１．逆に外部コンピュータから制御コマンドを送っている状態で、ジョイスティックを操作した場合はどのような挙動になるでしょうか。
- ジョイスティックの操作は無視され、外部コンピュータの制御コマンドが反映され続ける
- 操作している時だけジョイスティックの入力が有効になり、ジョイスティックが定位置に戻るとすぐに外部コンピュータの制御コマンドが反映される
- ジョイスティックの操作が有効になり、それ以降外部コンピュータの制御コマンドは無視される

２．コードを読んでみたところ、制御コマンドをシリアル通信で送ることができればROS2だろうと、Ubuntu22.04だろうと制御可能という認識をしました。この認識はあっているでしょうか。

３．シリアル変換モジュールなどは要らないでしょうか。RS232Cケーブルを挿すだけで（弊社のPCにRS232Cケーブルは無いため、何かしらのUSB変換ケーブルを使って）ハード的な回路は完成という理解でよろしいでしょうか。

s-katsu, 2022-10-17:
>１．逆に外部コンピュータから制御コマンドを送っている状態で、ジョイスティックを操作した場合はどのような挙動になるでしょうか。
>

基本的には機体側のジョイスティックの操作量が、制御コマンドよりも優先されますので、下記の挙動となります。

>操作している時だけジョイスティックの入力が有効になり、ジョイスティックが定位置に戻るとすぐに外部コンピュータの制>御コマンドが反映される

ただし、通信仕様書の3.2.4のSetJoystickコマンドに記載の通り、
https://github.com/WHILL/whill_control_system_protocol_specification/blob/master/WHILL_Control_System_Protocl_Specification.pdf

U0の箇所を0/1どちらに設定するかで外部コマンドを優先するかも設定変更可能です。

[image]

ROS1でいうと、この設定値になります。
https://github.com/WHILL/ros_whill/blob/melodic-devel/src/whill/whill_commands.cpp#L75

>２．コードを読んでみたところ、制御コマンドをシリアル通信で送ることができればROS2だろうとPythonだろうと、>Ubuntu22.04上だろうと制御可能という認識をしました。この認識はあっているでしょうか。

はい、その認識であっております。
結局はROSなど各種ライブラリはシリアル通信を実行しているだけです。

>３．シリアル変換モジュールなどは要らないでしょうか。RS232Cケーブルを挿すだけで（弊社のPCにRS232Cケーブルは無>いため、何かしらのUSB変換ケーブルを使って）ハード的な回路は完成という理解でよろしいでしょうか。

はい、PC(USB)と接続する場合はRS232 - シリアル変換のケーブルをご使用ください。
下記の記事などが参考になるかと思います。

https://note.com/katsushun89/n/n3784991a6143

keita1523, 2022-10-25:
@s-katsu 
ありがとうございます。
2つ前のs-katsuさんに送って頂いたコメントについて質問させてください。
> 人が前にいないときは、機体のジョイスティック操作量をそのまま外部コンピュータからジョイスティック操作量として設定
> 人が前にいるときは、外部コンピュータからのジョイスティック操作量を0として設定

以下2点質問です。
- 「機体のジョイスティック操作量をそのまま外部コンピュータからジョイスティック操作量として設定」とはどういうことでしょうか。気概のジョイスティックの操作量がなぜ外部コンピュータを通るのかが不思議に思いました。

- 機体のジョイスティックを操作すると制御値は直接機体へ飛ばされるが、その状態で外部コンピュータから制御値を送信すると外部コンピュータの制御値のみ反映されるようになるという認識です。これはあっているでしょうか。

s-katsu, 2022-10-26:
> 「機体のジョイスティック操作量をそのまま外部コンピュータからジョイスティック操作量として設定」とはどういうことでしょうか。気概のジョイスティックの操作量がなぜ外部コンピュータを通るのかが不思議に思いました。

例えば、ROS1では機体のジョイスティックの操作量はこのtopicでpublishされます。

https://github.com/WHILL/ros_whill#statesjoy-sensor_msgsjoy

これをそのまま、
https://github.com/WHILL/ros_whill#controllerjoy-sensor_msgsjoy
でsubscribeされる値に反映させると外部コンピューターを経由していますが、機体のジョイスティックの操作量で動作することができます。
勿論外部コンピューターを経由するためのシステムディレイが追加されますのでその影響は考慮する必要はあります。

keita1523, 2022-10-27:
つまり下図で言う赤い矢印の流れでデータを回し、「制御値計算ノード」でジョイスティックのデータを使うか、停止させるかを判定すればよいということでしょうか。

[image]

s-katsu, 2022-10-28:
> つまり下図で言う赤い矢印の流れでデータを回し、「制御値計算ノード」でジョイスティックのデータを使うか、停止させるかを判定すればよいということでしょうか。

はい、そうです。
停止させたい場合は/controller/joyの出力をゼロなどにしてもらえばよいです。

keita1523, 2022-11-08:
追加で質問です。
[ros_whill](https://github.com/WHILL/ros_whill)に入っているxacroを表示させたのですが、全く車椅子の形をしていないように見えます。
そもそもそういうものではないということでしょうか？
WHILL CRモデルを再現したものが記述されていると思っておりました。

[image]

s-katsu, 2022-11-14:
1に関しては、機体のモデルを表したxacroではございません。

2に関してはホイール半径とトレッド幅(タイヤ間の長さ）はこちらの値を参照ください。
https://github.com/WHILL/ros_whill/blob/c61f6e8936a8f0dd4db3ffac99603d537b8972c1/src/whill/WHILL.h#L122-L123

closed 2022-12-29 by github-actions[bot]