WHILL/mrp-support #42: TXD-RXD間の電圧について

useuse01, 2022-10-28:
毎々お世話になっております。

ModelCRの問い合わせでこちらを紹介されたので、以下内容教えていただきたく。

・車いす制御用ケーブルコネクタのBATTの3端子の電源側にヒューズがついているという認識であっていますでしょうか？
・車いす制御用ケーブルコネクタのTXD-RXD間の電圧を測定した所、DCで約28Vだったのですが正しい電圧でしょうか？

車いす本体に改造は行っておらず、電圧はケーブルコネクタの負荷側を端子受けして測定しております。
他にスイッチング電源など接続しておらず、PLC、TP、ブレーカをBATT-GND間に接続しております。

以上、お手数ですがよろしくお願いいたします。

s-katsu, 2022-10-28:
お問い合わせありがとうございます。

> ・車いす制御用ケーブルコネクタのBATTの3端子の電源側にヒューズがついているという認識であっていますでしょうか？
はい、その認識であっております。

> ・車いす制御用ケーブルコネクタのTXD-RXD間の電圧を測定した所、DCで約28Vだったのですが正しい電圧でしょうか？

通信のラインはRS232Cになっており、その仕様の通りの電圧差が出ています。
TXDとRXDでGNDに対してプラスとマイナスの電位差を持つのでTXDとRXDの電位差をみると大きな電圧に見える可能性はあります。
RS232Cで通信できないなども問題があるのでしょうか？

useuse01, 2022-10-28:
ご連絡ありがとうございます。

set powerのｺﾏﾝﾄﾞをRS232Cで送信しているのですがmodelCRの電源がONにならず、考えられる原因あれば
教えていただきたく。
ｺﾏﾝﾄﾞの内容は以下の通りです。
AF 02 02 01 AE(XOR)

通信設定は以下の通りに行っております。
通信速度:38400
パリティビット:無し
データ長:8
ストップビット:2

また、制御機器はPLC通信ﾓｼﾞｭｰﾙ(FC6A-SIF52)を使用しております。
PLC側でﾓﾆﾀするとTXDでコマンドを送信完了しているようなのですが、modelCRが反応しないといった状況です。
WHILL プロトコル仕様書の8 コネクター側TXDをPLC側のRXD、コネクター側RXDをPLC側TXDに接続して
おります。
通信ﾓｼﾞｭｰﾙのGNDはバッテリーコネクタのGNDと接続しております。

以上、お手数ですがご意見いただきたくお願いいたします。

s-katsu, 2022-10-28:
詳細ありがとうございます。

まずSetPower（ON）のバイト列は
0xAF, 0x03, 0x02, 0x01, 0xAF
となります。
詳細は仕様書ご確認ください。
https://github.com/WHILL/whill_control_system_protocol_specification/blob/master/WHILL_Control_System_Protocl_Specification.pdf

またSetPower(ON)は10ms間隔で2回送信していただく必要がありますので、その点をお試しいただきたいです。

もう一点、問題の切り分けとして、ノートPCなどにRS232C-シリアル変換ケーブルを繋いでいただき、
pythonなどのライブラリで動作するか確認いただくことも可能です。
そちらも検討いただけると幸いです。
https://github.com/WHILL/pywhill

非公式となりますが、こちらの記事などもご参照ください。
https://note.com/katsushun89/n/n3784991a6143

https://note.com/katsushun89/n/n6287998d48a6

closed 2022-12-13 by github-actions[bot]