WHILL/mrp-support #44: WHILLのキーボード操作のROSによる実装について

ryo0616, 2022-11-03:
現在，私はWHILL model CRのキーボード操作をROSにより実装したいと考えています．しかし，サイトなどで調べてもその方法がよくわからない状況です．
そこで，WHILLのキーボード操作をROSで実装する方法があれば教えていただきたいです．

お手数をおかけしますが，よろしくお願いします．

s-katsu, 2022-11-03:
Issueの再起票ありがとうございます。

もう少し具体的に何を行いたいか教えていただきたいのですが、

> WHILL model CRのキーボード操作をROSにより実装したい

とおっしゃるのは、Model CRとPCをケーブルで接続した状態で、
PCのキーボード操作（例えば矢印キーで）機体を前後左右などに走行させたいということでしょうか？

例えばROSのTutorialで亀をキーボードで動かすものがあるのですが、その亀をModel CRに置き換えたようなイメージでしょうか？
http://wiki.ros.org/ROS/Tutorials/UnderstandingTopics

ryo0616, 2022-11-03:
返信ありがとうございます．
Model CRとPCをケーブルで接続した状態で、
PCのキーボード操作（例えば矢印キーで）機体を前後左右などに動かさせたいということで間違いないです．

s-katsu, 2022-11-03:
こちらを利用してキーボード操作→cmd_velのpublishができます。

http://wiki.ros.org/stdr_simulator/Tutorials/Teleop%20with%20teleop_twist_keyboard

```
$ rosrun teleop_twist_keyboard teleop_twist_keyboard.py cmd_vel:=/whill/controller/cmd_vel
```

上記コマンドの通り、topic名をros_whill側に合わせて、変更してください。
https://github.com/WHILL/ros_whill#controllercmd_vel-geometry_msgstwist

機体がいきなり動くので最初は後輪をジャッキアップ（地面につかない状態に）してお試しください。

ryo0616, 2022-11-04:
上記のコマンドを実行したところ
"Waiting for subscriber to connect to /whill/controller/cmd_vel"
と表示され続けるのですが、待ち続ければよいのでしょうか？

ryo0616, 2022-11-04:
キーボードで操作することができました。ありがとうございました。最後に、上記のもの利用して、キーボード以外での操作が可能なのかどうか教えていただきたいです。

closed 2022-12-22 by github-actions[bot]