WHILL/mrp-support #47: pywhillを利用して電源をｵﾝｵﾌする方法について

useuse01, 2022-12-23:
毎々お世話になっております。

以前TXD-RXD間の電圧について#42という題で問い合わせをさせていただいておりました。
その後、pywhillを利用して、以下サンプルコードを使用させていただいてPCとWhiLL間で通信を試みましたが
上手くいかず、お手数ですがお知恵をお貸しいただきたく。

import time
from whill import ComWHILL

def main():
    whill = ComWHILL(port='COM5')
    while True:
        time.sleep(3)
        whill.send_power_off()
        time.sleep(3)
        whill.send_power_on()
        print('ON')

if __name__ == "__main__":
   main()

コマンドプロンプトにてwhill_on.pyという名前で実行しております。
ポートはデバイスマネージャーで確認してCOM5を記載しました。
ｼﾘｱﾙ、USB変換機でTXDの信号出力時にランプが点灯するのですが、WhiLLの電源に変化
ありませんでした。
コマンドプロンプトで実行した後はエラーは出力せず、"ON"は周期的に出力されるようでした。

以上、お手数お掛けいたしますが、よろしくお願いいたします。

useuse01, 2022-12-26:
すみません、追加情報なのですが、
シリアル-USB変換器とWHiLLとの間は御社のコネクタ変換ケーブルを使用しております。
https://whill.inc/jp/model-cr/accessories

以上、お手数ですがよろしくお願いいたします。

s-katsu, 2022-12-26:
お問い合わせありがとうございます。

もう少し状況を教えていただきたいですので下記の点ご回答いただけますでしょうか。

---
##### 確認1

 Model CRの本体の電源ボタン押して、電源が入った状態で
`whill.send_power_off()`
を実行しても電源オフできないのでしょうか？

---

##### 確認2

Model CRの本体の電源が入っていない（本体の電源ボタンでオフにされた）状態で
`whill.send_power_on()`
を実行して電源が入りますか？（アームの先のLEDが点灯しますか？）

---

##### 確認3
pythonの問題か切り分けするために、Power Shell上で動作するこちらのツールで電源ON,OFFできるかご確認いただけますでしょうか。

https://github.com/whill-labs/cr_connection_checker

使用方法は[README](https://github.com/whill-labs/cr_connection_checker/blob/main/README.md)に記載しております。

下記の通りdevice.jsonのデバイス名をCOM5に変更すれば通信できるかと思います。

https://github.com/whill-labs/cr_connection_checker#device%E5%90%8D%E3%81%AE%E8%A8%AD%E5%AE%9A
```
{
    "Device": "COM5"
}
```

---
##### 確認4

もし可能でしたらPCからModel CRまでの接続の様子を画像などで教えていただけますでしょうか。

useuse01, 2022-12-29:
ご連絡ありがとうございます。

確認1
電源オフできませんでした。

確認2
電源オンできませんでした。

確認3
[image]
batファイルを実行するとpowershellでこのような表示がされ、.exeファイルも生成されていないようでした。
(実行するファイルはutilフォルダ内のremane_file.batで合ってますでしょうか？)

確認４
接続は以下の通りに行っております。

[image]
[image]
[image]

以上、お手数ですがよろしくお願いいたします。

useuse01, 2022-12-29:
ご連絡ありがとうございます。

確認1
電源オフできませんでした

確認2
電源ONできませんでした

確認3
[image]
batファイル実行するとこのような表示が出ました。.exeファイルも生成されていないようです。
エラーの原因がわからず、お手数ですが想定される要因わかれば教えていただきたく

確認４
接続は以下の通りに行っております。
(ヒューズが飛んでおり、現在はｹｰﾌﾞﾙで短絡させていますが、ヒューズが飛ぶ以前でも電源ONできませんでした)
[image]
[image]
[image]

以上、お手数ですがよろしくお願いいたします。

s-katsu, 2023-01-10:
> 確認3
> batファイル実行後
> batファイル実行するとこのような表示が出ました。.exeファイルも生成されていないようです。
> エラーの原因がわからず、お手数ですが想定される要因わかれば教えていただきたく

ご確認ありがとうございます。

batファイルはcr_connection_checkerをcr_connection_checker.exeにrenameするだけですので、もし上手くいかない場合は、手動でファイル名変更して.exeを追加、powershell上で実行を試していただけますでしょうか。

useuse01, 2023-01-13:
ご連絡ありがとうございます。

cr_connection_checkerというファイルが見当たらないのですが(cr_connection_checker-mainというフォルダはあるのですが...)
zipファイルをダウンロードして展開するやり方は間違っているのでしょうか？
初歩的な質問で申し訳ないのですが、教えていただきたくお願いいたします。

s-katsu, 2023-01-13:
https://github.com/whill-labs/cr_connection_checker/releases/tag/0.1.0
のRelease/0.1.0にあります

[cr_connection_checker_windows_amd64.zip](https://github.com/whill-labs/cr_connection_checker/releases/download/0.1.0/cr_connection_checker_windows_amd64.zip)

をダウンロードしてご利用ください。

useuse01, 2023-01-16:
上記のツールを接続しましたので以下画像を添付いたします。

[image]

[image]

[image]
[image]

ｼﾘｱﾙ、USB変換機の都合でdevice.jsonのデバイス名をCOM４に指定しております。
WHiLLの電源をONしてもモニタ出来ていないようでした。

以上、お手数ですがよろしくお願いいたします。

s-katsu, 2023-01-17:
ご確認ありがとうございます。
そうしますと、故障の可能性がございますので、ご購入された際にやりとりしていた営業の窓口までお問い合わせいただけますでしょうか？
お手数をおかけして申し訳ございませんが、よろしくお願いいたします。

useuse01, 2023-02-22:
上記件、承知いたしました。
[image]
お手数ですが添付写真のヒューズの型式を教えていただけないでしょうか。
お手数ですがよろしくお願いいたします。

s-katsu, 2023-03-03:
返信遅れまして申し訳ありません。
こちらになります。
https://www.amon.jp/products2/detail.php?product_code=1262

useuse01, 2023-03-06:
上記件、確認いたしました。
ご連絡ありがとうございました。

s-katsu, 2023-04-17:
こちらCloseさせていただきます。

closed 2023-04-17 by s-katsu