WHILL/mrp-support #48: Whillの片流れ防止機能について

sazanami-sazanami, 2023-01-18:
以前購入したWhill model CRを使用しております。

ご質問なのですが、
貴社HP（[https://whill.inc/jp/model-c）](https://whill.inc/jp/model-c)
において、
Whill modelCには片流れ防止機能があると記載があります。
片流れ防止機能について、以下の内容にてご質問させていただきたく存じます。
１）modelCRにも搭載されていますでしょうか。
２）modelC/CRともに、モデル初期から搭載されていますでしょうか。
３）片流れ防止はどのようにして行われていますでしょうか。例えば、ジャイロセンサを利用してヨー軸角速度を0にする、あるいはIMU情報にカルマンフィルタに通して姿勢推定を行い、機体姿勢を制御する、といった制御方法についてです。
４）片流れ防止機能はいつ作動するでしょうか。例えば、ロール軸方向にどれくらい傾いたときか、あるいは平面でも（３）の方法の制御が常に作動しているのか。また、進路によって作動するかどうか変わるか。すなわち、直進しているときだけなのか、直進＋旋回（ジョイスティックを右上／左上に倒したような状態）でも作動するか。
５）片流れ防止機能の性能はどれくらいでしょうか。すなわち、どれくらいの横断勾配まで対応しており、どれくらいの直進性が得られるか。

以上ご回答いただけますと幸いです。

KazumichiShirai, 2023-01-20:
前提として、他社比較の性能であり、sazanami-sazanami様の期待ほどの性能はない可能性が高いようにです。
申し訳ございません。
基本的には、モーターの制御を路面の傾斜によらず精度良く制御しているものだと考えて頂けたらと思います。

> １）modelCRにも搭載されていますでしょうか。

搭載されています。

> ２）modelC/CRともに、モデル初期から搭載されていますでしょうか。

搭載されています。

> ３）片流れ防止はどのようにして行われていますでしょうか。例えば、ジャイロセンサを利用してヨー軸角速度を0にする、あるいはIMU情報にカルマンフィルタに通して姿勢推定を行い、機体姿勢を制御する、といった制御方法についてです。

そこまでのセンサフュージョンはしておりません。
基本的には、モーターの制御を路面の傾斜によらず精度良く制御しているものです。 (多少工夫はしておりますが、詳細は控えさせていだきます。)

> ４）片流れ防止機能はいつ作動するでしょうか。例えば、ロール軸方向にどれくらい傾いたときか、あるいは平面でも（３）の方法の制御が常に作動しているのか。また、進路によって作動するかどうか変わるか。すなわち、直進しているときだけなのか、直進＋旋回（ジョイスティックを右上／左上に倒したような状態）でも作動するか。

基本的には常時作動しています。

> ５）片流れ防止機能の性能はどれくらいでしょうか。すなわち、どれくらいの横断勾配まで対応しており、どれくらいの直進性が得られるか。

明確な測定値はお出しできません。申し訳ございません。

以上が回答になります。ご参考になれば幸いです。
ご期待に応える回答でなければ、申し訳ございません。

sazanami-sazanami, 2023-01-24:
ご回答ありがとうございます。参考にさせて頂きます。

closed 2023-01-24 by sazanami-sazanami