WHILL/mrp-support #50: シリアル通信の不具合に関して

sato-shu, 2023-04-05:
お世話になっております。

以前WHILL Model CRを購入させていただいたのですが、WHILL本体とのシリアル通信ができないという現象が発生し、こちらでは対処不可能と判断したため、原因調査または修理をお願いしたく、ご連絡させていただきました。
以下に詳細をまとめましたので、ご確認いただきたく思います。

-Model名：Model CR
-シリアルNo ：C_2105097_JP
-発生事象等(お申し出内容)：
WHILLを組み立て後に付属のケーブルを用いてPCと接続したところ、疎通を確認することができませんでした。
<こちらで実施したこと>
•疎通の可否を確認するために以下のサイトを参考にバイナリデータのやり取りを実施いたしましたが、正常の動作を確認することはできませんでした。正常な場合はバイナリーデータが表示されますが、それらが表示させませんでした。
https://qiita.com/KazumichiShirai/items/a023d3a7e5f9cd81b405
•通信以外の走行等の動作は正常に動くことを確認済みです。
•疎通が確認できなかったWHILLとは別に同じtypeのWHILLをご購入させていただいているのですが、そちらでは正常な接続を確認できております。この時使用したケーブルやPCは、どちらも同じものを使用しています。
•RS232CからUSBに変換するケーブルとして、以下を使用しました。
UGREEN USB シリアル変換ケーブル RS232 USB 9ピン 変換 シリアルケーブル D-sub9ピン Windows Mac OS両対応 1m https://amzn.asia/d/belG4qi
•WHILL本体およびケーブル等に外傷は見られません。

以上お手数ですが、内容をご確認いただき、ご連絡していただけますと幸いです。
よろしくお願いいたします。

s-katsu, 2023-04-05:
ご連絡ありがとうございます。

>•疎通が確認できなかったWHILLとは別に同じtypeのWHILLをご購入させていただいているのですが、そちらでは正常な接続を確認できております。この時使用したケーブルやPCは、どちらも同じものを使用しています。

他の機体では通信出来ているということですと、機体の故障の可能性が高いです。

申し訳ございませんが、こちらのgithub repogitoryではModel CRの機能などのご質問を受けつけておりまして、故障や修理に関しましては、WHILLホームページ側からお問い合わせをお願いします。

お手数ですがこちらの修理についてというページからご連絡お願い致します。
https://whill.inc/jp/support/repair

sato-shu, 2023-04-05:
ご返答いただきありがとうございます。

承知いたしました。そちらで申請させていただきます

s-katsu, 2023-04-28:
こちらCloseさせていただきます。

closed 2023-04-28 by s-katsu