WHILL/mrp-support #51: ファームウェアアップデートの方法

kuogata, 2023-04-10:
githubの方で，ファームウェアアップデートをしないと記載の機能が使えないとのことで，WHILL Model Cのファームウェアアップデートをしたいのですが，やり方が分かりません．
お手数ですが，ご教授頂けますでしょうか．

s-katsu, 2023-04-11:
ご連絡ありがとうございます。
Model CRのアップデートは営業経由で日程調整させていただきます。
お手数おかけしますが、購入した際にやりとりしていた営業窓口までご連絡お願いいたします。

kuogata, 2023-04-11:
B投失礼します．
Bjをいた方へメ`ルをしましたが，メ`ルが届きませんでした．
こちらの方は今はそちらにいらっしゃらないのでしょうか．

赤X
***@***.***

ご_Jの程，よろしくおいいたします．
尾形
________________________________
差出人: Shunsuke Katsumata ***@***.***>
送信日r: 2023年4月11日 13:59
宛先: WHILL/Model_CR_Technical_Support ***@***.***>
CC: 尾形邦裕 ***@***.***>; Author ***@***.***>
件名: Re: [WHILL/Model_CR_Technical_Support] ファ`ムウェアアップデ`トの方法 (Issue #51)

ごBjありがとうございます。
Model CRのアップデ`トはIU由で日程{整させていただきます。
お手数おかけしますが、入したHにやりとりしていたI口までごBjおいいたします。

s-katsu, 2023-04-17:
申し訳ありませんが、メールでご連絡つかないようでしたらこちらの窓口からご連絡お願いいたします。

お手数ですがこちらの修理についてというページからご連絡お願い致します。
メールが届かなかった旨記載お願いします。
https://whill.inc/jp/support/repair

kuogata, 2023-04-17:
承知しました．
ありがとうございます．
取り急ぎ，御礼まで．

尾形
________________________________
差出人: Shunsuke Katsumata ***@***.***>
送信日r: 2023年4月17日 13:38
宛先: WHILL/Model_CR_Technical_Support ***@***.***>
CC: 尾形邦裕 ***@***.***>; Author ***@***.***>
件名: Re: [WHILL/Model_CR_Technical_Support] ファ`ムウェアアップデ`トの方法 (Issue #51)

申しUありませんが、メ`ルでごBjつかないようでしたらこちらの口からごBjおいいたします。

お手数ですがこちらの修理についてというペ`ジからごBjおい致します。
メ`ルが届かなかった旨dおいします。
https://whill.inc/jp/support/repair

s-katsu, 2023-06-21:
こちらCloseさせていただきます。

closed 2023-06-21 by s-katsu