WHILL/mrp-support #52: ros2_whillについて

takamatsuait, 2023-04-11:
お世話になっております。
私は大学の研究室で学生としてROSベースで御社のWHILL Model CRの作製を行っております。
今回新たにROS2を元にロボットを作製しようと考え以下のパッケージをインストールしました。
https://github.com/whill-labs/ros2_whill
しかし、以下のようなエラーが出てしまいました。
現在使っていますubuntuバージョンはubuntu20.04、ROS2バージョンはROS foxyです。何か導入方法について分かることがありましたご教授お願い致します。お忙しいところ恐れ入りますが何卒よろしくお願いいたします。
[image]

s-katsu, 2023-04-11:
ご連絡ありがとうございます。

申し訳ありませんが、
whill-labs以下のリポジトリはWHILL社では公式にサポートしておりません。（有志でのサポートになります）
ros2_whillリポジトリ側でissue立ててご連絡お願いします。
ただ、foxy自体はサポートするようにメンテ出来ていない可能性があります。

takamatsuait, 2023-04-11:
ご返信ありがとうございます。
承知いたしました。お忙しい中ご迷惑をおかけして申し訳ございません。

s-katsu, 2023-04-13:
こちらはCloseさせていただきます。
お問い合わせありがとうございます。