WHILL/mrp-support #54: SetVelocity Commandの使用方法について

user0101sd, 2023-06-19:
お世話になります。

現在、pythonのset_speed_profile関数を用いて、speed_profileの変更をしようと思っています。
サンプルプログラムのcr_example6_copy_speed_profile.pyは問題なく動作するのですが、
任意のspeed_profileに設定しようとすると、うまく設定できません。

現状、例えば下記のようなコードで試してみています。
 whill.set_speed_profile(4,10,10,10,10,10,10,10,10,10)

どのように変更すればよいかご教示いただければ幸いです。
よろしくお願いいたします。

s-katsu, 2023-06-21:
お問い合わせありがとうございます。

> 現状、例えば下記のようなコードで試してみています。
> whill.set_speed_profile(4,10,10,10,10,10,10,10,10,10)

set_speed_profile関数で設定できる範囲外の値が設定されている可能性があります。

通信仕様書に設定可能な範囲が記載されていますのでその中に全てのパラメータが収まるかたちで再度試していただけますでしょうか。
https://github.com/WHILL/whill_control_system_protocol_specification/blob/master/WHILL_Control_System_Protocl_Specification.pdf

[image]

例えば、F_D1は40〜160が設定可能なレンジになります。

user0101sd, 2023-06-22:
ご回答ありがとうございました。
試したところ、問題なくspeed profileを変更できました。
単純なミスで大変失礼いたしました。

解決できましたので、本件closeさせていただきます。

closed 2023-06-22 by user0101sd