WHILL/mrp-support #55: SetVelocity commandで後方指令時の速度の急激な減少について

useuserxx, 2023-07-19:
現在、C#で.NetFramework上で仕様書をもとに、シリアル通信でWHILL Model CRを制御しています。SetVelocity commandで制御する際にbackの値を-30で指令を出し続けている場合に、Whill state dataでモータの速度を取得しています。以下のように安定した動作を行っています。

[image]

しかし、ある程度したときに以下のように急激に速度が落ちてしまいます。その後速度は元に戻り、安定した走行を行っていますが、この急激に速度が落ちる動作が定期的に起きます。

[image]

このような動作中でもWhill state dataでのERRORは0のままで、バック以外をSetVelocity commandで行う場合はこのような動作は起こっていません。PCからの指令の送信間隔は200ms以内です。SetJoystick commandを使用する場合にこの動作は起こっていませんでした。この動作をどうすれば防げるでしょうか。

初歩的なミスかもしれませんが、教えていただきたいです。
よろしくお願いいたします。

[image]

s-katsu, 2023-10-04:
こちら返信が非常に遅れて申し訳ありません。
ファームウェアの修正が必要となりますので、修正検討中です。
ファームウェアの更新に関してはまた追ってご連絡させていただきます。
申し訳ございません。

s-katsu, 2023-10-31:
返信遅くなりましたが、本不具合の修正版のFirmwareがリリース可能となりました。
ご迷惑おかけして申し訳ありません。

購入の際にやりとりされていた営業担当にご連絡いただくか、こちらからご連絡いただければ、Firmwareの更新の対応致します。
https://whill.inc/jp/whill-contact#content-form-corporate-customer-rental

GeorgeMandokoro, 2026-09-13:
その後、いかがでしょうか。
ご返信をいただけていないため、本 Issue はいったんクローズいたします。

解決していない場合や、あらためてご不明な点が出てきた場合は、
このスレッドにコメントいただければ再開いたします。
新しいご質問として Issue を作成いただいても構いません。

お問い合わせいただき、ありがとうございました。

closed 2026-09-13 by GeorgeMandokoro