WHILL/mrp-support #56: arduino用ライブラリ whill-sdk-arduinoについて

transport246, 2023-10-04:
私はGithubからwhill-sdk-arduino-masterのzipファイルをインストールして､arduino ideの「.zip形式のライブラリをインストール」からライブラリを追加しました。しかしWHILL.hをインクルードしたプログラムを作りコンパイルしようとした所､WHILL.cppの中でインクルードされているSoftwareSerial.hが存在しないというエラーが出て実行出来ません。SoftwareSerial.hはArduino ideにデフォルトであるはずなのですが､どうすれば問題無く実行できるようになるでしょうか。Arduino ideは最新版の2.2.1を使っています。
[image]

s-katsu, 2023-10-04:
お問い合わせありがとうございます。

SoftwareSerial.hはArduino環境（使用しているボード）に依存しますのでそちらをご確認いただければ幸いです。
恐れ入りますが、弊社のサポートの範疇外となります。

また参考情報となりますが、esp32用にHardwareSerial.hでアクセスするesp32ブランチも用意しています。
https://github.com/WHILL/whill-sdk-arduino/tree/esp32

差分
https://github.com/WHILL/whill-sdk-arduino/commit/931f5487993efe1077deabc47c39b6edfc47fea1

参考記事(HardwareSerial使用時）
https://qiita.com/KatsuShun89/items/69222ce96ff3debcba11

GeorgeMandokoro, 2026-09-13:
その後、いかがでしょうか。
ご返信をいただけていないため、本 Issue はいったんクローズいたします。

解決していない場合や、あらためてご不明な点が出てきた場合は、
このスレッドにコメントいただければ再開いたします。
新しいご質問として Issue を作成いただいても構いません。

お問い合わせいただき、ありがとうございました。

closed 2026-09-13 by GeorgeMandokoro