WHILL/mrp-support #57: 24V供給電源について

kentanaka1203, 2023-10-30:
Model CRのSpecificationsに"供給電源 24V (1Ax3ピン), GND"といった記載があり
搭載している負荷に電源供給できそうなように見受けられるのですが、この
- 端子位置
- コネクタ詳細
- その他利用にあたっての注意事項
を教えて頂けませんでしょうか?

s-katsu, 2023-10-31:
お問い合わせありがとうございます。

端子・コネクタの情報はこちらのドキュメント参照ください。（コメントから既にご参照いただいているようではありますがリンクは貼らせていただきます）
[whill_control_system_protocol_specification](https://github.com/WHILL/whill_control_system_protocol_specification)

該当部分の抜粋画像を貼付します。
BATTと記載されているピンがBatteryのラインになります。
[image]

> その他利用にあたっての注意事項
こちらは用途によりますが、ドキュメント記載の通り、バッテリーの電圧ラインを直接出力したものになりますので、
消費電流3A以下にしていただく必要があるのと、一定の電圧が出力されるわけではないので必要に応じてDCDCなどを使用してください。

kentanaka1203, 2023-10-31:
@s-katsu  ありがとうございます!

closed 2023-10-31 by kentanaka1203