WHILL/mrp-support #58: 直進指令のみを与えているが，左に曲がってしまう現象について

hosoik, 2023-12-25:
大学の研究でWHILL Model CRを使用しています．
直進の速度指令のみを与えているのですが，10m進むと約20cmほど左に曲がってしまう現象が起こっています．
ROSを使用して直進のみの速度指令を与えているので左右への速度指令は入っていません．
また，床面も様々な場所で検証を行ったため，床が斜めだからという理由も考えられないと思います．
最近この現象に気付いたため購入時からこの状態なのかはわかりません．

シリアルナンバーはC_2001083_JPです．

修理をせずこちら側で対応できることがあれば教えていただきたいです．
また，修理が必要な場合は修理をお願いいたします．

s-katsu, 2024-01-10:
お問い合わせありがとうございます。
回答送れて申し訳ありません。

> 10m進むと約20cmほど左に曲がってしまう
こちらですが、回答難しいのですが、まっすぐ進むことを補償しているわけではないため、直進指令だけを与えた場合、ある程度距離を進むと左右にズレが生じる可能性はあります。

例えばタイヤの微妙なサイズの個体差などにもよって変わって来ることがあるため、故障かの断定は難しいです。
初期位置の向きによっても同様に左右にずれていくことは考えられます。

Model CRの制御においては、こういった誤差を許容した制御をしていただくのをご検討いただけると助かります。

hosoik, 2024-01-11:
ご返信ありがとうございます。
承知いたしました。お忙しい中ご迷惑をおかけして申し訳ございません。

本件closeさせていただきます．

closed 2024-01-11 by hosoik