WHILL/mrp-support #59: 一時停車中のWhill model CRの状態について

DISK49, 2024-01-10:
現在私は研究にて御社のWhill model CRを使用しています．
研究にて加速度センサーを付け加速度を測定を行っているのですが，一時停車後，発進するときの加速度が遅くなっていることに気づきました．
その要因として御社のWhill model CRに停車中はアイドリングストップが搭載されていることにより，一時停車後の発進の加速度が遅くなっているのではと考えました（停車直後と発進する際に発生する「カコン」という音がそうではないかと思いました）．それについて自分なりに調べてみたところ掲載されているところが見当たらなかったのでお問い合わせをさせていただきました．

既に記載ないしは既出であるないようでしたら申し訳ありませんが，お時間がある時にご返事いただけると幸いです．よろしくお願いします．

s-katsu, 2024-01-12:
ご連絡ありがとうございます。

> 加速度の測定を行っているのですが，一時停車後，発進するときの加速度が遅くなっていることに気づきました．

この現象をもう少し具体的に説明いただけると助かりますが、
記載いただいた情報からコメントさせていただきます。

> （停車直後と発進する際に発生する「カコン」という音がそうではないかと思いました）．

この「カコン」という音は電磁ブレーキをロックしている音になります。
後輪に前方に赤いバーがありますが、それが電磁ブレーキの手動解除スイッチになり、
停車時にはこの電磁ブレーキをロックするため、そのときに音がしています。

ですので、停車後に再度走行するときに加速が遅いというのは、電磁ブレーキがちょうどロックされたタイミングで再度走行しようとすると、ブレーキロック→解除する時間がかかるので、加速が遅れるのではないかと推測されます。

DISK49, 2024-01-12:
ご返信ありがとうございます．
説明が不足してしまい申し訳ありません．停車後の加速度が遅くなっていることについて詳しく記載させていただきます．

私は加速度センサーを取り付け，Whill model CRの最大加速度を測定するため実験を行っていました．
実験方法は，停車状態から最大入力を入力したときの電動車いすの加速度を逐次測定し，その測定した値の中の最大値を取得していく方法です．

しかし，この実験をしていく中で方法の違いによって最大加速度の値が違うことに気づきました．違いが出た実験方法についてそれぞれ記載させていただきます．

・一時停車状態から，発進するときの最大加速度を測定
・一時停止をせず急ブレーキで止まった後，すぐに急発進したときの最大加速度の測定

この両方の測定方法をしたとき，一時停止していたときの最大加速度より止まった後，すぐに急発進したときの方が最大加速度が大きく出ました．このようなことから最大加速度の違いが何故生まれるのか疑問に思ったためお問い合わせさせていただきました．

ご返信いただいた内容から推察しますと，一時停止の際に電磁ブレーキによるロック後，再度走行する際にロック解除の時間がかかるため動き出すまでの時間がかかるのか，もしくは電磁ブレーキが完全に解除させる前に動き出すので加速が緩やかになるなるのか判断することが出来ませんでした．ご教授いただければ幸いです．

お忙しいと思いますが，お時間がある際にご返信いただければ幸いです．よろしくお願いします．

KazumichiShirai, 2024-01-26:
ご連絡ありがとうございます。

> この実験をしていく中で方法の違いによって最大加速度の値が違うことに気づきました

> 一時停止の際に電磁ブレーキによるロック後，再度走行する際にロック解除の時間がかかるため動き出すまでの時間がかかるのか，もしくは電磁ブレーキが完全に解除させる前に動き出すので加速が緩やかになるなるのか判断することが出来ませんでした

最大加速度とは、model CRの停止状態から所定の速度に達するまでの加速度を加速度センサーで測定して、その測定値の最大値のことでしょうか？
もしそうでしたら、最大加速度はモーターが動き出す時間には依存しないので、ロック解除の時間等は無関係かと思います。

DISK49, 2024-01-29:
ご返信ありがとうございます．

> 最大加速度とは、model CRの停止状態から所定の速度に達するまでの加速度を加速度センサーで測定して、その測定値の最大値のことでしょうか？

はい，その通りでございます．ただ，本実験を行うにあたり速度入力ではなく，ジョイスティックの入力で行っております．

電磁ブレーキが問題ではないと考えますと，私では思いつかないのですが，他に考えうる原因はありますでしょうか？

お手数をおかけしますが，お時間がある際にご返信いただければ幸いです．よろしくお願いします．

KazumichiShirai, 2024-01-31:
ご連絡ありがとうございます。

> 電磁ブレーキが問題ではないと考えますと，私では思いつかないのですが，他に考えうる原因はありますでしょうか？

加加速度制御しているので、すぐに急発進したときの加速度は少しオーバーシュートしているのかもしれません。

ちなみに、最大加速度が違うことで何かお困りでしょうか？困りごとを教えていただけたら、何か解決策をご提案できるかもしれません。

DISK49, 2024-02-09:
ご返信ありがとうございます．
ご連絡が遅くなってしまい申し訳ありません．

> ちなみに、最大加速度が違うことで何かお困りでしょうか？困りごとを教えていただけたら、何か解決策をご提案できるかもしれません。

最大加速度が違いましたので平均値などを用いることで特に論文などには問題ありませんでした．
しかし，卒業論文の執筆にあたり原因の詳細がもし分かればと思いご連絡させていただきました．

質問に答えていただき，ありがとうございました．

closed 2024-03-14 by s-katsu