WHILL/mrp-support #60: 後進中の一時停止動作について

883reon, 2024-01-17:
ros_whillでWHILL Model CRを制御しています．ros_whillに"/whill/controller/cmd_vel"で後進する一定の速度司令値を配信すると，2，3秒周期で一瞬停止する動作が入ります．ros_whillには速度司令値が正しく届いており，またros_whillからWHILLへも正しく速度司令値を送信できていましたので，WHILL本体側に原因があるのではないかと考えております．
なお，前進時や，ジョイスティックで操作する際はこのような挙動を示しません．
「https://github.com/WHILL/Model_CR_Technical_Support/issues/55」と同じ問題でしたら申し訳ないのですが，対処方法のほどご教授いただきたく思います．
よろしくお願いいたします．

以下に簡単に入力と出力をprintしたものを示します．
「Publish~」がros_whillにPublishした速度司令値，「ros_whill～」がros_whillのsetSpeed内のx,y値，「odom～」がオドメトリとして得られる実際WHILLが走行している速度値です．
===============================================
[INFO] [1705462241.402279]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.420223]: odom linear.x: -0.195, angular.z: -0.013
ros_whill linear:-179, angular:0
[INFO] [1705462241.421399]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.441923]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.441947]: odom linear.x: -0.192, angular.z: 0.000
ros_whill linear:-179, angular:0
ros_whill linear:-179, angular:0
[INFO] [1705462241.451236]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.463952]: odom linear.x: -0.203, angular.z: -0.018
[INFO] [1705462241.481457]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462241.490640]: odom linear.x: -0.195, angular.z: 0.013
ros_whill linear:-179, angular:0
[INFO] [1705462241.501773]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.521623]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462241.522963]: odom linear.x: -0.199, angular.z: -0.027
ros_whill linear:-179, angular:0
[INFO] [1705462241.542749]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.559957]: odom linear.x: -0.192, angular.z: -0.027
ros_whill linear:-179, angular:0
[INFO] [1705462241.572695]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.582001]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462241.587084]: odom linear.x: -0.156, angular.z: 0.013
ros_whill linear:-179, angular:0
[INFO] [1705462241.612785]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.624220]: odom linear.x: -0.133, angular.z: -0.053
[INFO] [1705462241.632278]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462241.651067]: odom linear.x: -0.159, angular.z: 0.027
[INFO] [1705462241.651540]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
ros_whill linear:-179, angular:0
[INFO] [1705462241.672400]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.683309]: odom linear.x: -0.126, angular.z: -0.027
[INFO] [1705462241.692468]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
ros_whill linear:-179, angular:0
[INFO] [1705462241.714074]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.726351]: odom linear.x: -0.113, angular.z: -0.027
ros_whill linear:-179, angular:0
[INFO] [1705462241.743300]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.752724]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462241.758854]: odom linear.x: -0.076, angular.z: 0.013
ros_whill linear:-179, angular:0
[INFO] [1705462241.771833]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.791471]: odom linear.x: -0.057, angular.z: -0.018
[INFO] [1705462241.801893]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462241.813346]: odom linear.x: -0.027, angular.z: -0.000
ros_whill linear:-179, angular:0
[INFO] [1705462241.822074]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462241.841713]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.850597]: odom linear.x: -0.023, angular.z: -0.013
[INFO] [1705462241.852030]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462241.872878]: odom linear.x: -0.013, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462241.882910]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.894886]: odom linear.x: -0.013, angular.z: -0.000
ros_whill linear:-179, angular:0
[INFO] [1705462241.902389]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.921315]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462241.922175]: odom linear.x: -0.007, angular.z: 0.000　　　　<-ここ付近で停止しています
ros_whill linear:-179, angular:0
[INFO] [1705462241.941454]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.954220]: odom linear.x: -0.007, angular.z: -0.000
ros_whill linear:-179, angular:0
[INFO] [1705462241.972019]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462241.975954]: odom linear.x: 0.003, angular.z: -0.013
[INFO] [1705462241.982365]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462241.999390]: odom linear.x: 0.013, angular.z: -0.000
[INFO] [1705462242.013056]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.023348]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.028302]: odom linear.x: 0.020, angular.z: 0.000
[INFO] [1705462242.046522]: odom linear.x: 0.009, angular.z: 0.018
[INFO] [1705462242.052394]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.068046]: odom linear.x: 0.027, angular.z: -0.027
[INFO] [1705462242.072237]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.085114]: odom linear.x: 0.010, angular.z: 0.013
[INFO] [1705462242.091969]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.106699]: odom linear.x: 0.000, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.112023]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462242.113641]: odom linear.x: -0.035, angular.z: -0.013
[INFO] [1705462242.141283]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.145713]: odom linear.x: -0.070, angular.z: 0.013
[INFO] [1705462242.151202]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
ros_whill linear:-179, angular:0
[INFO] [1705462242.171662]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462242.177574]: odom linear.x: -0.109, angular.z: -0.013
ros_whill linear:-179, angular:0
[INFO] [1705462242.191376]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462242.209459]: odom linear.x: -0.126, angular.z: -0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.221904]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462242.236266]: odom linear.x: -0.132, angular.z: -0.027
ros_whill linear:-179, angular:0
[INFO] [1705462242.242421]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.252449]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462242.273540]: odom linear.x: -0.142, angular.z: -0.013
[INFO] [1705462242.282226]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.300239]: odom linear.x: -0.148, angular.z: -0.009
ros_whill linear:-179, angular:0
[INFO] [1705462242.302198]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.321523]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462242.326950]: odom linear.x: -0.152, angular.z: -0.027
[INFO] [1705462242.341331]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.348569]: odom linear.x: -0.162, angular.z: 0.013
[INFO] [1705462242.372320]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.380824]: odom linear.x: -0.185, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.382213]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.412079]: Publish linear.x: -0.200, angular.z: 0.000
[INFO] [1705462242.417665]: odom linear.x: -0.195, angular.z: -0.013
[INFO] [1705462242.421632]: Publish linear.x: -0.200, angular.z: 0.000
ros_whill linear:-179, angular:0
[INFO] [1705462242.444669]: odom linear.x: -0.212, angular.z: 0.000
================================================

s-katsu, 2024-01-26:
返信遅れて申し訳ありません。
#55  と同じ対応になり、機体のファームウェアのアップデートが必要となります。
下記の通りご依頼お願いいたします。

>購入の際にやりとりされていた営業担当にご連絡いただくか、こちらからご連絡いただければ、Firmwareの更新の対応致します。
https://whill.inc/jp/whill-contact#content-form-corporate-customer-rental

883reon, 2024-01-27:
ご返信ありがとうございます．
承知いたしました．

ご対応いただきありがとうございました．
本件closeさせていただきます．

closed 2024-01-27 by 883reon