WHILL/mrp-support #61: WHILLの電源がつかない

egalab2023, 2024-01-18:
現在私は研究にて御社のWhill model CRを使用しています．
研究にてマイコンを使用し，WHILLの制御を行っています．
WHILLとマイコンを自作RS232ケーブルを用いて通信を行っているのですが，RX/TX,GNDの３つのうちGNDが外れていて，
マイコンからWHILL側のGNDピンのさし間違えが原因なのかWHILLの電源がつかなくなってしまいました．
同じような原因でWHILLの電源がつかなくなってしまったと考えられる人のサイトを添付させていただきます．
https://gachimoto.hatenablog.com/entry/2018/12/09/221914
こちらで確認したようにモーターコントローラのトランシーバー及びマイコンが故障していたそうです。 原因は、TX/RXとバッテリーのピンが接触し、トランシーバーに定格以上の電圧が加わったためだと考えられます．
[image]

egalab2023, 2024-01-19:
どのように対処すればいいでしょうか?
よろしくお願いします。

s-katsu, 2024-01-26:
> こちらで確認したようにモーターコントローラのトランシーバー及びマイコンが故障していたそうです。 原因は、TX/RXとバッテリーのピンが接触し、トランシーバーに定格以上の電圧が加わったためだと考えられます．

返信遅れて大変申し訳ございません。
こちらですがご記載頂いている通り、故障していると推測されます。

お手数おかけしますが、故障や修理に関しましては、WHILLホームページ側からお問い合わせいただくか、
購入時の担当営業にご連絡をお願いいたします。

ホームページとしては、お手数ですがこちらの修理についてというページからご連絡お願い致します。
https://whill.inc/jp/support/repair

closed 2024-03-14 by s-katsu