WHILL/mrp-support #64: WHILL Model CRのバッテリ温度低下のエラーコードについて

nakajimawai, 2024-04-30:
お世話になっております。

WHILL Model CRで研究をさせていただいている者です。
現在、以下に示すような走行できない状態になっているのですが、原因は何と考えられるのでしょうか？

1.  常温下（15℃）でバッテリを装着し、電源をつけると「バッテリ温度低下のため、速度が制限されます」という音声が流れる。

2.  ジョイスティックを傾け、動かそうとすると自動でブレーキがかかり、停止してしまう（速度設定は1）。

3. ノートPCとのRS232Cによる制御信号の通信で動かそうとしても自動でブレーキがかかる。

4.  電源を切り、バッテリを再度装着し、再起動しても1, 2と同じ状態。

5.  1時間後にバッテリを再度装着し、再起動しても1, 2と同じ状態。

補足事項：

・本日の日中は正常に動いており、最後に電源を落としたのは17:00です。そして18:30に再度使おうとしたところ上記の状態になりました。

・17:00に電源を落とした時のバッテリ残量は90程度でしたが、18:30に起動した際はバッテリ残量はなぜか100と表示されていました。

・バッテリ本体の残量確認ボタンを押すと、ランプは5個点灯していました。

・購入時期は2023年1月頃であり、購入から約1年ほど経過しています。今まで充電した回数は20回程度です。

nakajimawai, 2024-05-01:
[image]

追記させていただきます。

本体から外しての充電時は

充電器：赤点灯
バッテリ本体：バッテリゲージ無点灯

という状態でした（添付写真参考）。

s-katsu, 2024-05-01:
ご連絡ありがとうございます。

頂いた情報からはバッテリーの単体故障のように考えられます。
・満充電表示のバッテリーに充電器をつないでも、赤ランプの点灯状態が続くこと。
・バッテリー温度低下を測るサーミスタはバッテリー内部に設置されており、バッテリーからの情報で温度低下を判断していること。
他に機体をお持ちなのであれば、バッテリーをスワップしていただいて、症状が機体にとどまるのか、別の機体に移るのかで切り分けていただくことも可能です。
 
機体側に不具合がある可能性もございますが、お客様の改造の状態によっては我々では判断いたしかねる部分がありますのでご了承ください。また、機体全体をお預かりして不具合要因を確認することもできますので、必要に応じてご用命ください。

お手数おかけしますが、故障や修理に関しましては、WHILLホームページ側からお問い合わせいただくか、
購入時の担当営業にご連絡をお願いいたします。
ホームページとしては、お手数ですがこちらの修理についてというページからご連絡お願い致します。
https://whill.inc/jp/support/repair

nakajimawai, 2024-05-01:
お忙しい中教えていただきありがとうございます。

機体は1台のため、スワップによる確認はできませんが、新しいバッテリの購入を検討します。

もしバッテリの故障だった場合、「再発防止のため」と「バッテリ購入の判断材料のため」、その原因を知りたいのですが、何が原因で故障するのでしょうか？

s-katsu, 2024-05-09:
申し訳ありませんが、本GitHub repositoryは研究開発モデルであるModel CRの技術的な使用方法についての問い合わせのためのものになりますので、ここで故障について回答はできません。

バッテリーはModel Cのものと共通であり、そういったお問い合わせは下記の通りお願いいたします。
お手数おかけしますがよろしくお願いいたします。

> お手数おかけしますが、故障や修理に関しましては、WHILLホームページ側からお問い合わせいただくか、
購入時の担当営業にご連絡をお願いいたします。
ホームページとしては、お手数ですがこちらの修理についてというページからご連絡お願い致します。
https://whill.inc/jp/support/repair

closed 2024-07-10 by nakajimawai