WHILL/mrp-support #65: AE-TTL-232Rとブレッドボードの配線の接続方法。

Euclid-7, 2024-06-14:
私は、写真の白い箱の中にあるブレッドボードの回路がどうなっているかよく分かっておりません。
そもそも回路に関しては全く知識がありません。
白い箱の中にあるブレッドボードから出る4本の配線(赤と黒と白と黄)のそれぞれをAE-TTL-232-Rのどの箇所に
繋げばよいか分かりません。繋ぐ場所を間違えて使うと壊れたりするでしょうか。
また、4本の配線が適切な場所に繋がれていれば、AE-TTL-232RとWindows10をUSBコードで接続して、
私のパソコン(Windows10PRO)の中にVirtual Box（仮想マシン）を立ち上げ、
ROSを使ったシリアル通信を行う事は可能でしょうか。
添付ファイルは閲覧可能ですか。

上記の質問に答えていただけない場合、どこに問い合わせればよいか教えていただきたいです。
よろしくお願いいたします。

https://github.com/WHILL/Model_CR_Technical_Support/assets/147054619/718a238a-d80b-4375-8897-211eb5d6a2e0

[image]
[image]
[image]

s-katsu, 2024-06-17:
お問い合わせありがとうございます。

申し訳ありませんが、記載頂いている、”白い箱"は弊社製品ではありませんので、お答えすることは出来ません。

弊社コネクタのRS232Cの接続関しましては、8章を参照ください。

https://github.com/WHILL/whill_control_system_protocol_specification/blob/master/WHILL_Control_System_Protocol_Specification.pdf
[image]

PCにつないでUSB経由のシリアル通信を行う場合はこちらの記事を参照ください。
こちらに書かれているものを揃えていただくのが無難かと思います。

https://note.com/katsushun89/n/n3784991a6143

GeorgeMandokoro, 2026-09-13:
その後、いかがでしょうか。
ご返信をいただけていないため、本 Issue はいったんクローズいたします。

解決していない場合や、あらためてご不明な点が出てきた場合は、
このスレッドにコメントいただければ再開いたします。
新しいご質問として Issue を作成いただいても構いません。

お問い合わせいただき、ありがとうございました。

closed 2026-09-13 by GeorgeMandokoro