WHILL/mrp-support #70: WHILL Model CRの加速度センサについて

yf11290618, 2024-07-26:
お世話になっております。
大学の研究でWHILLを使用している者です。
御社の製品WHILL Model CRについて質問がございます。
WHILLの仕様書だと思われる ”WHILL Control System Protocol Specification” の
7.1節に記載されている内容についてRev2とRev9で加速度センサの方向が異なっておりました。
私どもが使用しているWHILLの加速度センサはどちらに当てはまりますでしょうか。

使用している製品
　・モデル：WHILL Model CR（2017年購入）
　・シリアル番号：C_170****_JP

以上です。
お手数おかけしますが、ご教示いただけますよう、よろしくお願いいたします。

[WHILL Control System Protocol Specification_Rev2.pdf](https://github.com/user-attachments/files/16388325/WHILL.Control.System.Protocol.Specification_Rev2.pdf)
[WHILL Control System Protocol Specification_Rev9.pdf](https://github.com/user-attachments/files/16388326/WHILL.Control.System.Protocol.Specification_Rev9.pdf)

GeorgeMandokoro, 2024-07-31:
お問い合わせいただきありがとうございます。

製造年月に関わらず、加速度センサの方向は、[Rev9](https://github.com/WHILL/whill_control_system_protocol_specification/blob/master/WHILL_Control_System_Protocol_Specification.pdf)の記載が正しい方向となります。
（記載ミスが見つかった時点（Rev5）で、通信仕様書を訂正させていただきました。ご迷惑をおかけし申し訳ございませんでした）

以上、よろしくお願いいたします。

yf11290618, 2024-08-02:
ご返信いただきありがとうございます。
承知いたしました。

本件Closeさせていただきます。

closed 2024-08-02 by yf11290618