WHILL/mrp-support #71: whillの電源が入らない

hagi-okuda, 2024-09-25:
お世話になっております。
model CR2のwheeled robot baseの電源が入らなくなりました
本体についているリモコン、無線のカギ、スマホ操作、ubuntuからのシリアル通信で起動、操作ができないことを確認しました
バッテリーは付属のACアダプタの緑ランプが点灯しているので満タンです
どのように対処すればよろしいでしょうか？
以上、よろしくお願いいたします。

GeorgeMandokoro, 2024-09-27:
お問い合わせいただきありがとうございます。

お手数をおかけしますが、バッテリーを充電している状態で、バッテリー側面にある「バッテリーLED」の状態をご連絡いただけますでしょうか？
満充電では、緑色となります。

[image]

hagi-okuda, 2024-09-27:
返信ありがとうございます。
充電を始めるとバッテリーLEDが赤色に点灯したり、青色に点滅します
しばらくするとバッテリーLEDは消灯します
[image]

GeorgeMandokoro, 2024-09-27:
ご確認いただきありがとうございます。

バッテリー故障の可能性がございますので、大変お手数をおかけしますが、
ロボット台車のシリアルナンバー（本体ジョイスティック横のラベルにある「23」から始まる11桁）をメールにてお知らせいただけますでしょうか？

- メールアドレス（担当直通）
mrp.contact@whill.inc

以降は、メールにてご対応させていただきたいと思いますので、よろしくお願いいたします。

closed 2024-10-03 by hagi-okuda