WHILL/mrp-support #72: Model C2 ロボット台車の速度追従性能と車両仕様に関して

s-azumi, 2024-10-01:
Model C2ベースのロボット台車を[ros2_whill](https://github.com/whill-labs/ros2_whill)を利用して制御しています。
性能や仕様を確認していた際に気になる点がありましたので、以下2点についてご教示頂けないでしょうか。

**[速度追従性能について]**
SetVelocityコマンドで車両を直進させた際、速度入力に対して8%程低い数値で車速が安定するのですが、Model C2の制御性能としては妥当な数値でしょうか？
下図は青が速度入力[m/s]、赤が左右のモーター速度から計算した車両の速度（(left_motor_speed + right_motor_speed) / 2）です。
[image]

**[車両仕様について]**
オムニキャスター側を前、モーター側を後として左右を認識しているのですが、その場合にモーター速度が左右逆に出力されてしまっています。
また、車両の挙動とros2_whillリポジトリの[こちらのPR](https://github.com/whill-labs/ros2_whill/pull/5/files)を見る限り、旋回方向の正負が左手系に則している様なのですがこれは仕様でしょうか？（多くの場合は右手系が標準だと認識しています）
上記理由により左右のモーターもしくは配線が逆になってしまっている可能性を考えているのですが、車両仕様としての前後左右及び座標系はどれが正しい状態になるでしょうか？

GeorgeMandokoro, 2024-10-03:
お問い合わせいただきありがとうございます。
以下の通り、回答いたします。

> [速度追従性能について]
> SetVelocityコマンドで車両を直進させた際、速度入力に対して8%程低い数値で車速が安定するのですが、Model C2の制御性能としては妥当な数値でしょうか？

概ね意図通りの動作ではあります
（人を乗せた実験などを想定しており、加速度と上限速度をやや緩めている状態となっております）。
どのような実験・環境で利用されますでしょうか？
急峻な加速・速度追従が必要な場合は、個別にファームアップデートを実施させていただきますのでご連絡ください。

> [車両仕様について]
> オムニキャスター側を前、モーター側を後として左右を認識しているのですが、その場合にモーター速度が左右逆に出力されてしまっています。
> また、車両の挙動とros2_whillリポジトリのこちらのPRを見る限り、旋回方向の正負が左手系に則している様なのですがこれは仕様でしょうか？（多くの場合は右手系が標準だと認識しています）

「オムニキャスター側を前、モーター側を後として左右を認識している」は正しいです。

ros2_whillのインターフェースとしては、ROS慣例の右手系（進行方向がx軸、左方向がy軸、上方向がz軸の正方向です。回転方向は反時計周りが正(0～π[rad])、時計回りが負(0～-π[rad])）となっております。
言及いただいておりますPRによって正しいインターフェースになったという理解です。

すなわち、こちらのサンプルコマンドを実行すると、半時計回りに旋回することが正しい状態となります。

```ros2 topic pub -r 12 /whill/controller/cmd_vel geometry_msgs/Twist '{linear: {x: 0}, angular: {z: 0.785}}'```

また、他の確認手段としまして、本体のジョイスティックを操作してみていただけないでしょうか？
前に倒すと前進、後ろに倒すと後進、左に倒すと半時計回りの旋回、右に倒すと時計回りに旋回します。
なお、ジョイスティックはロボット台車の進行方向に対して左90度向きに付いております（電源ボタンがある側が正面）ので、その点はご注意ください。

お手数をおかけしますが、ご確認のほどよろしくお願いいたします。

s-azumi, 2024-10-04:
回答ありがとうございます、以下返答及び追加の質問となります。
長文となり恐縮ですが、何卒宜しくお願い致します。

> 概ね意図通りの動作ではあります
（人を乗せた実験などを想定しており、加速度と上限速度をやや緩めている状態となっております）。

元々加速度や上限速度を緩めている旨理解しました。
しかし、それらは車速が指示値より一定割合低くなってしまう事には影響していないと認識しているのですが、指示値通りの速度に追従するよう調整頂くことも可能という事でしょうか？（モーターのPID的な部分の調整になると思っています）

> どのような実験・環境で利用されますでしょうか？
急峻な加速・速度追従が必要な場合は、個別にファームアップデートを実施させていただきますのでご連絡ください。

前述したデータは車両のモーター側をローラー台車に載せた状態で、モーターが空転する状態で取得したデータになりますが、車両をローラー台から下ろし屋内の平地を走行させた際も同様の傾向が見られました。
また、車両にはセンサー数台とノートPCを載せているのみで、重量物は乗せておりません。

> ros2_whillのインターフェースとしては、ROS慣例の右手系（進行方向がx軸、左方向がy軸、上方向がz軸の正方向です。回転方向は反時計周りが正(0～π[rad])、時計回りが負(0～-π[rad])）となっております。
言及いただいておりますPRによって正しいインターフェースになったという理解です。

WHILL車両としては旋回方向が左手系の仕様になっており、ros2_whill側で右手系になるよう変換しているという事で理解しました。

> また、他の確認手段としまして、本体のジョイスティックを操作してみていただけないでしょうか？
前に倒すと前進、後ろに倒すと後進、左に倒すと半時計回りの旋回、右に倒すと時計回りに旋回します。

ご教示頂いた通りジョイスティックを操作した所、前後左右の全てが意図通りの挙動となりました。
また、ros2_whillのwhill_driverノードを起動し、サンプルコマンドを送信すると意図通り車両が左旋回しました。
しかし、whill_driverノードから出力されているright_motor_speed及びleft_motor_speedを確認すると、実際の車両とは出力が逆になってしまっていました。

以下のデータは上側が速度及び角速度の指示値、下側が車両から出力されている左右のモーター速度になります。
指示値は前進方向に0.2[m/s]、旋回方向は正の値で左旋回となっており、車両も左方向へ旋回しています。
この時、左方向へ旋回しているため車両右側のモーターの方が高い回転数となっているのですが、whill_driverから出力されているmotor_speedを確認すると、以下データの通りleft_motor_speedの方が速度が速くなってしまっている状態です。
（前提として、車両が前進した時、左モーターはマイナスの速度が出力されると理解しています）
[image]

[whill_driverのData set 1受信箇所](https://github.com/whill-labs/ros2_whill/blob/867f78b047c4c43073622d158a89149598c9b071/whill_driver/src/model_cr2/whill.cpp#L277-L279)と仕様書を照らし合わせても問題無い様に見えますので、車両側もしくは仕様書側のどちらかに間違いがあるのではないかと考えているのですが如何でしょうか？
[image]

GeorgeMandokoro, 2024-10-09:
ご確認いただきありがとうございます。
下記の通り回答いたします。

> しかし、それらは車速が指示値より一定割合低くなってしまう事には影響していないと認識しているのですが、指示値通りの速度に追従するよう調整頂くことも可能という事でしょうか？（モーターのPID的な部分の調整になると思っています）

はい。指示値通り速度追従できるようにいたします。ただし、およそ2%程度の誤差は生じることはありますのでご了承ください。

> [whill_driverのData set 1受信箇所](https://github.com/whill-labs/ros2_whill/blob/867f78b047c4c43073622d158a89149598c9b071/whill_driver/src/model_cr2/whill.cpp#L277-L279)と仕様書を照らし合わせても問題無い様に見えますので、車両側もしくは仕様書側のどちらかに間違いがあるのではないかと考えているのですが如何でしょうか？

ご指摘いただきありがとうございます。
調査しましたところ、ご指摘通りモーターの情報が左右逆となっておりました。ご不便をおかけし、申し訳ございません。
対策としまして、こちらで正しいファームウェアを作成し、WHILL Model CR2 / ロボット台車のユーザー様へ展開させていただきたいと考えておりますので、しばらくお待ちいただけますと幸いです。

なお、上記SetVelocityコマンド（指示値通りの速度追従）の改善につきましても、合わせて対応する予定です。

s-azumi, 2024-10-09:
ありがとうございます、速度追従の精度改善及び左右のモーター情報不整合の修正について対応頂けるとの事で承知しました。
お手数おかけしますが、対応の程宜しくお願い致します。

s-azumi, 2024-12-18:
こちらご連絡が大変遅くなりましたが、修正頂いたファームウェアを車両へ適用し、SetVelocityコマンドの改善及びモーター情報の整合性について修正されていることを確認致しました。

対応頂きありがとうございました、本件Closeさせて頂きます。

closed 2024-12-18 by s-azumi