WHILL/mrp-support #74: Model CR2の旋回に関する情報の取得方法に関する質問

WK-ME, 2024-10-24:
以下、2点質問があります。

① Model CR2の旋回中の実際の(もしくは推定の)旋回角度や角速度を取得する方法について
　 Model CR2について、コマンドで指定した旋回の角速度等ではなく、実際の(もしくは推定の)角速度等の情報を利用したいと考えています。
　 github掲載の「WHILL_Control_System_Protocol_Specification.pdf」を確認したのですが、
　「4.WHILL state data」の「4.2 Data Set0」で返される「TURN_ACCEL」/「TURN_DECEL」が、コマンドで指定した旋回加速度なのか、
　 何等かの方法で計測した実際の加速度かわかりませんでした。

　 そこで、
　　・前述の方法で実際の旋回加速度が取得できるのか
　　・できないのであれば、取得する方法はあるのか
　 についてご回答いただきたいです。

② 「SetSpeedProfile commnad」における「T_A1」、「T_D1」の2つのコマンドデータの分解能について
　　github掲載の「WHILL_Control_System_Protocol_Specification.pdf」内、「3.2.5 SetSpeedProfile commnad」では、
　　「T_A1」: ～ The range is (10～60).
    「T_D1」: ～ The range is (40～160).
    と記載されていますが、単位・分解能が読み取れませんでした。
　　上記2つのコマンドデータについて分解能をご回答いただきたいです。

以上、2点の質問について、ご回答いただければ幸いです。

GeorgeMandokoro, 2024-10-28:
お問い合わせいただきありがとうございます。
以下の通り、回答いたします。

> ○　「right_motor_angle」、「left_motor_angle」は具体的にはモータの出力軸の回転角度でしょうか。

はい。モータの出力軸の角度になります。単位は[rad]です。

> ○　「right_motor_speed」、「left_motor_speed」については、モータのどの箇所についての速度でしょうか。
> 　(Model CR2の旋回角速度の算出をしたく、上記モータ速度からModel CR2の旋回角速度へ変換できるのか知りたいためです。)

こちらは、角速度ではなく、車輪径を考慮した機体が進む速度に変換した値になります。単位は[km/h]です。
例えば、WHILL本体が1[km/h]で前進している場合、
・right_motor_speed：-1.0前後（車軸中心から向かって右回りが正）
・left_motor_speed：1.0前後
と表示されることが正しいです。
必要な長さ等を計測いただければ、旋回速度に変換いただけるかと思います。

なお、大変ご不便をおかけしますが、現在のファームウェアにおいて、
取得できるモーター情報（角度および速度（＝今回お問い合わせいただいている4項目））の左右が逆になっている不具合が見つかっております。
近日中に正しいファームウェアを公開予定ですので、しばらくお待ちいただけますと幸いです。

WK-ME, 2024-10-29:
ご回答ありがとうございます！

ファームウェアの不具合の件、了解しました。
確認になりますが、上記の例ですと、実際は
「right_motor_speed = -1.0前後
   left_motor_speed = 1.0前後」
   ではあるが、ファームウェアの不具合のため、
  「right_motor_speed = 1.0前後
   left_motor_speed = -1.0前後」
   と情報を渡してしまうという認識であっていますでしょうか。

GeorgeMandokoro, 2024-10-29:
ご連絡ありがとうございます。

> 確認になりますが、上記の例ですと、実際は
> 「right_motor_speed = -1.0前後
> left_motor_speed = 1.0前後」
> ではあるが、ファームウェアの不具合のため、
> 「right_motor_speed = 1.0前後
> left_motor_speed = -1.0前後」
> と情報を渡してしまうという認識であっていますでしょうか。

はい。こちら、ご認識の通りでございます。

WK-ME, 2024-10-29:
ご回答いただきありがとうございます！

新しいファームウェアの公開についてはよろしくお願いいたします！

GeorgeMandokoro, 2026-09-13:
その後、いかがでしょうか。
ご返信をいただけていないため、本 Issue はいったんクローズいたします。

解決していない場合や、あらためてご不明な点が出てきた場合は、
このスレッドにコメントいただければ再開いたします。
新しいご質問として Issue を作成いただいても構いません。

お問い合わせいただき、ありがとうございました。

closed 2026-09-13 by GeorgeMandokoro