WHILL/mrp-support #75: エラーコード「65」に関する問い合わせ

Masaya7707, 2024-10-25:
WHILL株式会社 技術サポートご担当者様
お世話になっております。
現在、貴社の車椅子ロボットを用いて屋外での自動走行実験を行っておりますが、エラーコード「65」が頻発しており、実験が進められない状況です。原因の特定および解決策のご教示をいただきたく、ご連絡差し上げました。

ロボットには自動走行の学習モデルを適用し、実験は主に屋外で行っています。以下に、エラーの発生状況を時系列でご報告いたします。

エラー発生状況（時系列）
9月30日
大学正門前および喫煙所前において走行実験を実施しました。どちらも平坦な舗装路で、特に障害物や段差はない環境です。この際は問題なく動作しました。なお、バッテリー残量が30％程度だったため、充電を行い翌日の実験に備えました。
10月1日
正門前で再度走行実験を試みた際、エラーコード「65」が発生しました。エラー対応として、一度バッテリーを取り外し、10秒程度待機してから再接続を行ったところ、一時的に動作は回復しましたが、間もなく同じエラーが再発し、以降の実験を進めることができませんでした。
10月2日
バッテリーを再度装着し、11階の廊下で手動走行テストを行ったところ、特に問題なく動作しました。その後、屋外で同様の走行実験を試みましたが、再度エラー「65」が発生しました。バッテリーを外して再接続する処置を行いましたが、エラーの発生間隔が徐々に短くなり、短距離走行ごとにエラーが再発する状態となりました。
10月7日
数日間バッテリーを外して保管した後、再度走行テストを実施しました。11階で手動操作により20〜30分程度走行させましたが、この時点ではエラーは発生しませんでした。続けて屋外で学習モデルを適用して実験を行いましたが、こちらもエラーなく走行できました。
10月13日
バッテリーを再度抜いて保管した後、車椅子ロボットで約10メートルの距離での走行実験を行いました。実験開始直後は正常に動作していたものの、計14回目の走行実験中に再度エラー「65」が発生し、その後はエラーが頻発する状態となりました。

問題点
これまでの経過から、エラー「65」が発生する原因が特定できておらず、現在、エラーが頻発し、実験の継続が困難な状況です。つきましては、原因や解決方法についてのご教示をいただけますでしょうか。
お忙しいところ恐れ入りますが、何卒よろしくお願い申し上げます。

GeorgeMandokoro, 2024-10-28:
お問い合わせいただきありがとうございます。
この度はご不便をおかけし申し訳ございません。

エラーコードより、内部のコネクタの故障の可能性があると考えられます。
お手数ですが、下記につきましても情報をいただけますと幸いです。

- 対象の機体は、「WHILL Model CR」または「WHILL Model CR2」のどちらでしょうか？
- 購入時期はいつごろになりますでしょうか？

以上、よろしくお願いいたします。

Masaya7707, 2024-10-29:
ご連絡いただきありがとうございます。
機体について、モデルは「WHILL Model CR」で間違いありません。
また、納品時期は2019年12月となります。

どうぞよろしくお願いいたします。

GeorgeMandokoro, 2024-10-30:
ご連絡ありがとうございます。

現状の復旧には修理（関連部品の交換等）が必要となります。
修理を希望される場合には、大変お手数をおかけしますが、
下記メールアドレスまでご連絡をいただけますでしょうか。

mrp.contact@whill.inc

以上、よろしくお願いいたします。

GeorgeMandokoro, 2026-09-13:
その後、いかがでしょうか。
ご返信をいただけていないため、本 Issue はいったんクローズいたします。

解決していない場合や、あらためてご不明な点が出てきた場合は、
このスレッドにコメントいただければ再開いたします。
新しいご質問として Issue を作成いただいても構いません。

お問い合わせいただき、ありがとうございました。

closed 2026-09-13 by GeorgeMandokoro