WHILL/mrp-support #76: ros2_whillのubuntu foxy用のdevelが無い件について

Euclid-7, 2024-11-07:
私は、Whill CR(2ではない方)を研究で利用しております。

・ros2_whill(humble-devel)パッケージの全てのコードをROS2 foxyで使えるように、私が自ら書き換えることは現実的に可能でしょうか。Whill CRに関する専門的な知識を持たない私ですが、いかがでしょうか。foxyでリリースされていない事に関しては、技術的に難しいところがあるのでしょうか。

・自ら書き換える上で、注意すべき点や、難しい部分を教えていただきたいです。

よろしくお願いいたします。

GeorgeMandokoro, 2024-11-11:
お問い合わせいただきありがとうございます。
下記の通り回答いたします。

> ・ROS2でWHILL CR(2ではない方)を動かすには、どうすればよいでしょうか。ros_whillパッケージ(ROS1専用)がありますが、これをROS2用に実装し直さなければならないでしょうか。

基本機能でしたら、ros2_whillをご利用いただけます。APIをご確認ください。
https://github.com/whill-labs/ros2_whill?tab=readme-ov-file#ros2-apis

なお、現状のros_whillにしかないAPIをご利用したい場合には、下記いずれかを行っていただく必要がございます。
- ros_whillをROS2用に実装し直す
- ros2_whillに必要なAPIを新規追加する

> ・ros2_whill(humble-devel)パッケージの全てのコードをROS2 foxyで使えるように、私が自ら書き換えることは現実的に可能でしょうか。Whill CRに関する専門的な知識を持たない私ですが、いかがでしょうか。foxyでリリースされていない事に関しては、技術的に難しいところがあるのでしょうか。

ROS2を扱ったご経験があれば技術的には問題ないと思われます。
Foxy対応版を公開していない理由は、現行版の公開を決めた時点でROS 2 FoxyがEOLになっていたためです。特に技術的な理由ではありません。

> ・自ら書き換える上で、注意すべき点や、難しい部分、前提として知っておくべき知識を教えていただきたいです

WHILL Model CRはRS232Cを使った通信を行います。
詳細は[WHILL_Control_System_Protocol_Specification.pdf](https://github.com/WHILL/whill_control_system_protocol_specification/blob/master/WHILL_Control_System_Protocol_Specification.pdf)をご覧ください。
また、ros2_whill(humble)は、Model CR/CR2共通の機能に対するライブラリとなっており、Model CR独自の機能には対応していないことに予めご注意ください（例えば、加速度センサの情報取得には対応していません）

> ・ros2_whillのhumble-develパッケージで、WHILL CR(2ではない方)を動かすことは可能ですか。

前述の通り、可能です。

以上、よろしくお願いいたします。

GeorgeMandokoro, 2026-09-13:
その後、いかがでしょうか。
ご返信をいただけていないため、本 Issue はいったんクローズいたします。

解決していない場合や、あらためてご不明な点が出てきた場合は、
このスレッドにコメントいただければ再開いたします。
新しいご質問として Issue を作成いただいても構いません。

お問い合わせいただき、ありがとうございました。

closed 2026-09-13 by GeorgeMandokoro