WHILL/mrp-support #77: WHILL Model CRのデータセット送受信に関して

kazuaki591, 2024-11-15:
貴社のWHILL model CRについてなのですが、現状マイコンを使ったコマンドの送信での速度調整やジョイスティックでの車いすの操作に成功しております。
しかし、車いすにStartSendingDataコマンドを送った後にデータセットが車いすから送信されません。この場合どのような不具合やミスが考えられるでしょうか。
StartSendingDataコマンドについては、こちらで確認して送信側に不具合がなかったと認識しております。
お答えいただけると幸いです。

GeorgeMandokoro, 2024-11-15:
お問い合わせいただきありがとうございます。

下記をご確認いただけますでしょうか？

- 本体の電源がONであること
- StartSendingDataコマンドのパラメータが範囲内であること
  - D0：0 or 1
  - S0：0 ~ 5

以上、よろしくお願いいたします。

kazuaki591, 2024-11-16:
本体の電源はオンで、コマンドデータはD0＝1、S0=0を入力しています。

GeorgeMandokoro, 2024-11-17:
ご連絡ありがとうございます。

WHILLからの受信に失敗している可能性があるかもしれません。

念のため確認なのですが、SetPowerコマンド（電源ON）にて、WHILLから応答は来ておりますでしょうか？
期待動作としましては、[通信仕様書](https://github.com/WHILL/whill_control_system_protocol_specification/blob/master/WHILL_Control_System_Protocol_Specification.pdf)の「5. Response Data」にあります4byteのデータが受信できていることになります。

以上、ご確認をお願いします。

kazuaki591, 2024-11-18:
SetPowerコマンドを送信してもデータを受信できませんでした。送信データはSetSpeedProコマンドによって移動ができているので送信はできていると思います。

GeorgeMandokoro, 2024-11-18:
ご確認いただきありがとうございました。

データ送信は問題なくできているようにお見受けします。
データ受信のみできていない症状であることより、RS232Cの受信側の設定の問題が高いと思われます。
お手数ですが、お使いのデバイスのRXピンにつきまして、物理的な接続やピン設定などハード/ソフト両面で、改めてご確認いただきますようよろしくお願いいたします。

kazuaki591, 2024-11-18:
こちらで使用しているデバイスの送受信は確認できました。
オシロスコープでRXDとTXDの波形を確認したところ、RXDの波形が動いた後もTXDの波形が全く変わらないこと、また車いすを分解してTXDとRXDをテスターで確認しましたが断線していないことが分かりました。

GeorgeMandokoro, 2024-11-18:
ご確認いただきありがとうございました。

念のため確認させていただきたいのですが、購入から一度もデータ受信ができていないということで理解合っておりますでしょうか？

kazuaki591, 2024-11-18:
先輩の卒論などを見るとデータセットを受け取れていないと出せないデータがあったので過去にはできていると思われます。

GeorgeMandokoro, 2024-11-18:
ご確認いただきましてありがとうございます。

以前、正常動作していたということでしたら、WHILL Model CRの基板が故障してしまった可能性があります。

大変お手数をおかけしますが、修理をご希望される場合には、下記窓口よりご連絡をお願いいたします。
[お問い合わせ・サポート](https://whill.inc/jp/whill-contact#content-form-private-customer-repair-request)

kazuaki591, 2024-11-18:
修理に出す場合、郵送する部分は基盤部分のみでも大丈夫でしょうか？

GeorgeMandokoro, 2024-11-18:
お手数をおかけしますが、修理内容に関しましては[窓口](https://whill.inc/jp/whill-contact#content-form-private-customer-repair-request)にてご相談くださいますようお願いいたします。

GeorgeMandokoro, 2026-09-13:
その後、いかがでしょうか。
ご返信をいただけていないため、本 Issue はいったんクローズいたします。

解決していない場合や、あらためてご不明な点が出てきた場合は、
このスレッドにコメントいただければ再開いたします。
新しいご質問として Issue を作成いただいても構いません。

お問い合わせいただき、ありがとうございました。

closed 2026-09-13 by GeorgeMandokoro