WHILL/mrp-support #79: WHILLのspeed profileについて

kakikuri, 2025-01-06:
お世話になります。
貴社のWHILL model CRの各速度モードの初期のSpeedProfileを教えていただきたいです。

GeorgeMandokoro, 2025-01-06:
お問い合わせいただきありがとうございます。
WHILL Model CRの出荷時のSpeedProfileの値は下記の通りとなります。

| S1 | F_M1 | F_A1 | F_D1 | R_M1 | R_A1 | R_D1 | T_M1 | T_A1 | T_D1 |
|:--:|:----:|:----:|:----:|:----:|:----:|:----:|:----:|:----:|:----:|
| 0  | 10   | 16   |  48  | 10   | 16   | 40   |  8   | 56   | 72   |
| 1  | 15   | 16   |  64  | 10   | 16   | 56   | 10   | 56   | 72   |
| 2  | 30   | 16   |  82  | 20   | 16   | 64   | 15   | 56   | 72   |
| 3  | 45   | 16   |  90  | 20   | 24   | 64   | 18   | 56   | 72   |
| 4  | 45   | 56   | 120  | 20   | 24   | 64   | 20   | 56   | 72   |
| 5  | 15   | 16   |  64  | 10   | 16   | 56   | 10   | 56   | 72   |

以上、よろしくお願いいたします。

kakikuri, 2025-01-07:
ご返信ありがとうございます。
追加でお聞きしたいのですが, F_M1, R_M1, T_M1の値の単位は0.1km/hですが, 他の値の単位についても教えていただきたいです。

GeorgeMandokoro, 2025-01-07:
ご連絡ありがとうございます。

その他のパラメータ（x_A1、x_D1）の単位につきましては、0.05[km/h/s]となっております。

以上、よろしくお願いします。

kakikuri, 2025-01-07:
角速度, 角加速度はどのように導出されていますか？

GeorgeMandokoro, 2025-01-07:
「角速度, 角加速度」につきまして、具体的にどこで示している数値になりますでしょうか？

kakikuri, 2025-01-09:
返信ありがとうございます
T_M1, T_A1,T_D1の数値の導出についてお聞きしたいです

GeorgeMandokoro, 2025-01-09:
ご連絡ありがとうございます。

それぞれ左右のモーターの合力から算出しております。
厳密な値が必要な場合、お手数ですがお使いの環境にて実測をお願いいたします。

GeorgeMandokoro, 2026-09-13:
その後、いかがでしょうか。
ご返信をいただけていないため、本 Issue はいったんクローズいたします。

解決していない場合や、あらためてご不明な点が出てきた場合は、
このスレッドにコメントいただければ再開いたします。
新しいご質問として Issue を作成いただいても構いません。

お問い合わせいただき、ありがとうございました。

closed 2026-09-13 by GeorgeMandokoro