WHILL/mrp-support #80: エラーコードの詳細について

s-azumi, 2025-01-10:
WHILL車両から受信可能なエラーコードについて、コード番号と対応するエラー詳細について公開されている資料はございますでしょうか？
WHILL車両を制御する上で、エラーコードの内容次第ではエラーハンドリング機能を実装したいと考えております。

またエラーコード詳細について公開可能な場合、本リポジトリにある通信仕様書と同じ様に公開して頂くことは可能でしょうか？

GeorgeMandokoro, 2025-01-13:
お問い合わせいただきありがとうございます。

公開しているエラーコードは、[こちらのFAQ](https://faq.whill.inc/jp/s/jp/Model-C-CK-C2-CK2-F%E3%81%AB%E3%82%A8%E3%83%A9%E3%83%BC%E3%82%B3%E3%83%BC%E3%83%89%E3%81%8C%E5%87%BA%E3%81%A6%E3%81%84%E3%81%BE%E3%81%99)に記載しておりますコードのみ（ディスプレイ表示が数字になっているエラー）となります。（Model CR2/ロボット台車/電装系キットは、＜Model C2/CK2/Fのエラーメッセージリスト＞をご参照ください）
上記以外のエラーコードが検出されて復旧しない場合は、お手数ですが別途ご連絡くださいますようお願いいたします。

申し訳ございませんが、エラーコードの詳細に関しまして通信仕様書に記載する予定はございません。

以上、よろしくお願いいたします。

s-azumi, 2025-01-17:
ありがとうございます、回答頂いたページの内容を参考にさせて頂きます。
また、エラーコード詳細については該当ページのみの記載という事で承知しました。

本件Closeさせて頂きます。

closed 2025-01-17 by s-azumi